mavlink/MAVSDK-Python #155: issue with MAVSDK-Python version 0.5.2

horstrand, 2020-01-22:
Hi,

I'm trying to run MAVSDK-Python on an embedded system. So far I have been using the Odroid XU4 board and the Jetson Nano.
When I follow the standard procedure for installing the package (pip3 install mavsdk), it strikes me as quite odd that the version that it is automatically installed in both boards is the 0.5.2. Whereas if I do the same in my PC, the version that is installed is the 0.4.0. With the latter I have no issue whatsoever, but I can't install that one on any of the boards.

If I install the version 0.5.2 in my PC I get the same error as in the boards, which is the following:

```File "/home/odroid/Documents/git/px4-mission-code/src/DSI/Drone.py", line 471, in _connect
    await self.system.connect(system_address="udp://:14540")
  File "/home/odroid/.local/lib/python3.6/site-packages/mavsdk/system.py", line 62, in connect
    self._start_mavsdk_server(system_address)
  File "/home/odroid/.local/lib/python3.6/site-packages/mavsdk/system.py", line 172, in _start_mavsdk_server
    stderr=subprocess.DEVNULL)
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: '/home/odroid/.local/lib/python3.6/site-packages/mavsdk/bin/mavsdk_server'
```

All systems are running Ubuntu 18.04.

I have first tried to give the mavsdk_server file the execution permissions, as per default it did not have them, with no success either.
And I have also tried to build both the mavsdk server and the Mavsdk-Python from source, again no success there.

I would appreciate very much if you could help me solve this issue.
Thank you in advance.

JonasVautherin, 2020-01-22:
Hello!

First of all, when I `pip install mavsdk`, I get mavsdk 0.6.1:

```
 % pip show mavsdk
Name: mavsdk
Version: 0.6.1
```

So I'm a bit surprised you get 0.5.2 and 0.4.0 respectively :thinking:.

A few questions:
* Which version of python are you running? You need 3.6+ (which should be fine on Ubuntu 18)
* Can you try to `pip3 install --user mavsdk` instead? More info [here](https://stackoverflow.com/questions/42988977/what-is-the-purpose-pip-install-user)
* If `--user` doesn't help, can you try to install it in a venv? That's how I usually do, here is how I would do in a /tmp folder:

```
mkdir /tmp/test_mavsdk_venv && cd /tmp/test_mavsdk_venv
python3 -m venv venv
source venv/bin/activate
pip install mavsdk
```

horstrand, 2020-01-22:
Hello Jonas,

Thank you for your fast reply!
First let me tell you that I also use virtual environments, but I was using another way to create them:

`virtualenv -p python3 venv`

which installs the MAVSDK-Python version 0.5.2. I now tried your way and it did install the version 0.6.1 in my PC, and if I do it without virtualenvs  (`pip3 install mavsdk`) I get the 0.6.1 too.

However when I try to replicate this steps in the Odroid board (ARM based) or in the raspberry pi I am getting the 0.5.2 version instead:

``` 
pi@raspberrypi:~ $ mkdir /tmp/test_mavsdk_venv && cd /tmp/test_mavsdk_venv
pi@raspberrypi:/tmp/test_mavsdk_venv $ python3 -V
Python 3.7.3
pi@raspberrypi:/tmp/test_mavsdk_venv $ python3 -m venv venv
pi@raspberrypi:/tmp/test_mavsdk_venv $ source venv/bin/activate
(venv) pi@raspberrypi:/tmp/test_mavsdk_venv $ pip install mavsdk
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mavsdk
  Downloading https://files.pythonhosted.org/packages/1f/ca/2b942796dc62686d10191d5a0784e47b9d782125a56473ba2be75fdbb11d/mavsdk-0.5.2-py3-none-any.whl (1.4MB)
    100% |████████████████████████████████| 1.4MB 175kB/s 
Collecting grpcio>=1.11.0 (from mavsdk)
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Downloading https://www.piwheels.org/simple/grpcio/grpcio-1.26.0-cp37-cp37m-linux_armv7l.whl (17.2MB)
    100% |████████████████████████████████| 17.2MB 13kB/s 
Collecting protobuf>=3.5.1 (from mavsdk)
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/02/ba/2d33a1ea34f672c088c08f0f8e70292ff770c7f0c2481690cae378d33d66/protobuf-3.11.2-py2.py3-none-any.whl (434kB)
    100% |████████████████████████████████| 440kB 404kB/s 
Collecting aiogrpc>=1.5 (from mavsdk)
  Downloading https://files.pythonhosted.org/packages/a3/61/00bb522858af1d0333316e3990c42027b2a2da8a14ff592616fc027bb658/aiogrpc-1.7-py3-none-any.whl
Collecting six>=1.5.2 (from grpcio>=1.11.0->mavsdk)
  Downloading https://files.pythonhosted.org/packages/65/eb/1f97cb97bfc2390a276969c6fae16075da282f5058082d4cb10c6c5c1dba/six-1.14.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools in ./venv/lib/python3.7/site-packages (from protobuf>=3.5.1->mavsdk) (40.8.0)
Installing collected packages: six, grpcio, protobuf, aiogrpc, mavsdk
Successfully installed aiogrpc-1.7 grpcio-1.26.0 mavsdk-0.5.2 protobuf-3.11.2 six-1.14.0
(venv) pi@raspberrypi:/tmp/test_mavsdk_venv $ pip show mavsdk
Name: mavsdk
Version: 0.5.2
Summary: Python wrapper for MAVSDK
Home-page: https://github.com/mavlink/MAVSDK-Python
Author: None
Author-email: None
License: UNKNOWN
Location: /tmp/test_mavsdk_venv/venv/lib/python3.7/site-packages
Requires: protobuf, grpcio, aiogrpc
Required-by: 
 ```
In the Odroid the python version is 3.6.9 as it is in my PC. In the rpi, as you could see there, is the 3.7.3.

julianoes, 2020-01-23:
In general, make sure to use `pip install --user --upgrade mavsdk` for the latest version.
I'm not sure why you are not getting the latest though, I can try to reproduce this on a RPi later.

horstrand, 2020-01-24:
Hi Julian, 

Thanks for your input.
We tried it in the rpi and still the same output though. Version 0.5.2 gets installed.

Since we are using virtual envs, not sure whether the `--user` flag makes much sense.

JonasVautherin, 2020-01-24:
I don't get why you can install on rpi with `pip install mavsdk`, though. Because we only support manylinux-x64 :thinking:.

Can you try to install from sources? In my venv, I usually do `pip install -e .` from the root of the repo. Then you need to run `mavsdk_server` manually (maybe you can get a binary that works on your RPi [here](https://github.com/mavlink/MAVSDK/releases/tag/v0.23.0)).

What RPi are you running? I'm interested in the cpu architecture (armv7? aarch64?).

julianoes, 2020-01-24:
Ok, I just tried this on Raspbian Buster on a Raspberry Pi 3 and I can reproduce the issue, so it also installs 0.5.2:

```
pi@raspberrypi:~ $ pip3 install --user mavsdk
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mavsdk
  Downloading https://files.pythonhosted.org/packages/1f/ca/2b942796dc62686d10191d5a0784e47b9d782125a56473ba2be75fdbb11d/mavsdk-0.5.2-py3-none-any.whl (1.4MB)
    100% |████████████████████████████████| 1.4MB 224kB/s 
Collecting grpcio>=1.11.0 (from mavsdk)
  Downloading https://www.piwheels.org/simple/grpcio/grpcio-1.26.0-cp37-cp37m-linux_armv7l.whl (17.2MB)
    100% |████████████████████████████████| 17.2MB 14kB/s 
Collecting aiogrpc>=1.5 (from mavsdk)
  Downloading https://files.pythonhosted.org/packages/a3/61/00bb522858af1d0333316e3990c42027b2a2da8a14ff592616fc027bb658/aiogrpc-1.7-py3-none-any.whl
Collecting protobuf>=3.5.1 (from mavsdk)
  Downloading https://files.pythonhosted.org/packages/02/ba/2d33a1ea34f672c088c08f0f8e70292ff770c7f0c2481690cae378d33d66/protobuf-3.11.2-py2.py3-none-any.whl (434kB)
    100% |████████████████████████████████| 440kB 354kB/s 
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio>=1.11.0->mavsdk) (1.12.0)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from protobuf>=3.5.1->mavsdk) (40.8.0)
Installing collected packages: grpcio, aiogrpc, protobuf, mavsdk
Successfully installed aiogrpc-1.7 grpcio-1.26.0 mavsdk-0.5.2 protobuf-3.11.2
```

JonasVautherin, 2020-01-24:
Maybe we changed something in the `setup.py` classifier after 0.5.2. The correct behavior is that it should not be available for RPi. However, I realize that what actually happens is that RPi apparently gets an older version.

I'll try to make a patch next week: it will publish a "pure" python version (i.e. that will not have `mavsdk_server` embedded). And therefore `mavsdk_server` will have to be started manually on those platforms that are not Linux-x64, Windows or macOS (when mavsdk is installed via `pip`).

horstrand, 2020-01-28:
Hi Jonas, 

Thanks again for the support.
We installed mavsdk-python from source, as you indicated, and used the mavsdk-server binaries you mentioned, more specifically the armv7 version.

This all works fine, and the version 0.6.1.  gets installed.

closed 2020-01-28 by horstrand

PCSmithy, 2020-04-14:
Hello all, 

I'm trying to install mavsdk on a NVidia Jetson TX2 to communicate with a pixhawk 4 and I'm getting the same mavsdk_server permission denied error that OP pointed out.

I'm working in a python3.8.2 venv, and when I `pip install mavsdk` it installs mavsdk 0.5.2.  

The Jetson TX2 is an arm64 architecture. Is my only option to build mavsdk from source, copy the arm64 mavsdk_sever from the releases and put it in `mavsdk/bin/mavsdk_server` or is there any way to use pip to install the correct version of mavsdk

I did try `pip install mavsdk==0.6.1`, but gave me an error saying there was no matching distribution...

Just wanted to make sure I wasn't missing something before I build mavsdk from source.

Thanks,

julianoes, 2020-04-15:
@PCSmithy you can use the binary for `mavsdk_server` that part is fine. Now you need to install the Python package as described here: https://github.com/mavlink/MAVSDK-Python#build-and-run-from-sources

hamishwillee, 2020-04-15:
@julianoes The [docs](https://mavsdk.mavlink.io/develop/en/getting_started/installation.html#python) say
[image]

If we are helping people with this, can we perhaps have a section in the doc on installing to ARM targets?

julianoes, 2020-04-16:
Yes we should, as soon as we get to it.

thomasderflieger, 2020-04-28:
Hey, wanted to connect my RPI3 with my pixhawk using MAVSDK and also just get the 0.5.2 Version when installing via pip. I am not very profficient with linux, nor building/installing etc. I tried building MAVSDK-Python from source. But what do i do now with  the mavsdk_server file?
And i guess i need this one: mavsdk_server_linux-armv7 

Sorry for the noob questions, i really like the idea of mavsdk after watching the auterion videos...
Thanks in advance.

PCSmithy, 2020-04-28:
Assuming you installed MAVSDK-Python correctly, there should be a directory in the MAVSDK-Python folder such as this:
```
~/Documents/MAVSDK-Python/mavsdk/bin # Mine is in my Documents folder for instance
```
That is where the binary of the mavsdk server needs to go. On my machine these were the commands I did:
```
sudo apt-get install wget # if you don't already have wget
cd ~/Documents/MAVSDK-Python/mavsdk/bin
wget https://github.com/mavlink/MAVSDK/releases/download/v0.24.0/mavsdk_server_linux-arm64 # change this for your specific architecture / release version
mv mavsdk_server_linux-arm64 mavsdk_server # Change the filename so it can be found
sudo chmod +x mavsdk_server # allow user to run, mine by default didn't have EXECUTE permissions
```
That should allow you to run the MAVSDK-Python examples (make sure you point the example code to the correct device for mavlink messages, ie UDP or Serial)

thomasderflieger, 2020-04-28:
Thanks for the quick reply!
Lookls like i was on the right path, but must have screwed up somewhere anyway. 
The only thing missing was the EXECUTE permission. I just tried the telemetry.py example. But i get a few error messages. I will keep tinkering with it tomorrow. 

```
pi@raspberrypi:~/MAVSDK-Python/examples $ python3 telemetry.py
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Task exception was never retrieved
future: <Task finished coro=<print_battery() done, defined at telemetry.py:18> exception=<_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.UNAVAILABLE
        details = "failed to connect to all addresses"
        debug_error_string = "{"created":"@1588112586.880813602","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3981,"referenced_errors":[{"created":"@1588112586.880803343","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":394,"grpc_status":14}]}"
>>
Traceback (most recent call last):
  File "telemetry.py", line 19, in print_battery
    async for battery in drone.telemetry.battery():
  File "/home/pi/MAVSDK-Python/mavsdk/generated/telemetry.py", line 2147, in battery
    async for response in battery_stream:
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/utils.py", line 138, in __anext__
    return await asyncio.shield(self._next_future, loop=self._loop)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/utils.py", line 126, in _next
    return next(self._iterator)
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/_channel.py", line 416, in __next__
    return self._next()
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/_channel.py", line 689, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.UNAVAILABLE
        details = "failed to connect to all addresses"
        debug_error_string = "{"created":"@1588112586.880813602","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3981,"referenced_errors":[{"created":"@1588112586.880803343","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":394,"grpc_status":14}]}"
............
```

In the meantime i also found this [thread](https://discuss.px4.io/t/talking-to-a-px4-fmu-with-a-rpi-via-serial-nousb/14119), explaining the installation of MAVSDK on a Raspberry Pi, as well as some more info [here](https://github.com/mavlink/MAVSDK/issues/997)

JonasVautherin, 2020-04-28:
> StatusCode.UNAVAILABLE

This hints that the connection to `mavsdk_server` failed. Which I find weird because above it says `Connected to mavsdk_server!`. Would you mind trying the following:

1. in `telemetry.py`, instead of 

```python
drone = System()
```

use:

```python
drone = System(mavsdk_server_address="localhost", port=50051)
```

2. run `./mavsdk_server -p 50051` manually on your system before you run `python3 telemetry.py`

This way you will actually get the output of `mavsdk_server` (which we currently drop when python starts it automatically :sweat_smile:). I would be interested in seeing the output of that :blush:

thomasderflieger, 2020-04-29:
Server Output:
```
pi@raspberrypi:~/MAVSDK-Python/mavsdk/bin $ ./mavsdk_server -p 50051
[12:21:18|Info ] MAVSDK version: 0.24.0 (mavsdk_impl.cpp:25)
[12:21:18|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:401)
[12:21:18|Info ] Server started (grpc_server.cpp:38)
[12:21:18|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:39)
[12:21:18|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:22)
```

telemetry.py output:
```
pi@raspberrypi:~/MAVSDK-Python/examples $ python3 telemetry.py
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Task exception was never retrieved
future: <Task finished coro=<print_battery() done, defined at telemetry.py:19> exception=<_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.UNAVAILABLE
        details = "Socket closed"
        debug_error_string = "{"created":"@1588155979.718846226","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1056,"grpc_message":"Socket closed","grpc_status":14}"
>>
Traceback (most recent call last):
  File "telemetry.py", line 20, in print_battery
    async for battery in drone.telemetry.battery():
  File "/home/pi/MAVSDK-Python/mavsdk/generated/telemetry.py", line 2147, in battery
    async for response in battery_stream:
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/utils.py", line 138, in __anext__
    return await asyncio.shield(self._next_future, loop=self._loop)
  File "/usr/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/aiogrpc/utils.py", line 126, in _next
    return next(self._iterator)
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/_channel.py", line 416, in __next__
    return self._next()
  File "/home/pi/.local/lib/python3.7/site-packages/grpc/_channel.py", line 706, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.UNAVAILABLE
        details = "Socket closed"
        debug_error_string = "{"created":"@1588155979.718846226","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1056,"grpc_message":"Socket closed","grpc_status":14}"
>
```
You want me to try anything else?
Else i will give it a new shot and follow the before mentioned thread.

JonasVautherin, 2020-04-29:
There are two things here:

1. `mavsdk_server` did not discover a drone. Is that expected?
2. Did you change [this line](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/telemetry.py#L9) in your example? Can you show me what you have there? It seems like MAVSDK-Python cannot reach `mavsdk_server` :thinking:

thomasderflieger, 2020-05-02:
Hm, it is expected to discover a pixhawk, running px4 on the serial port. 

The code:
   ```
 drone = System(mavsdk_server_address="localhost", port=50051)
 await drone.connect(system_address="serial:///dev/serial0:115200")
```

Thanks for the help by the way!

JonasVautherin, 2020-05-03:
Right. So when you run `mavsdk_server` manually, you should do:

```
./mavsdk_server -p 50051 serial:///dev/serial0:115200
```

It sounds a bit confusing, but `system_address` is only used when python starts `mavsdk-server` for you. Which is not the case when you set `mavsdk_server_address` to something that is not `None`.

maxwelllwang, 2020-05-14:
When I ran TakeoffAndLand.py I was also getting the message telling me to manually build mavsdk myself. Because I installed it using pip3. So I built it according to the directions in the C++ section of mavsdk.
Couple questions 
1. when i run ./mavsdk_server where do I do it? I've tried to run it from linux home and inside the mavsdk folder and it doesnt exist
2.  Also im using a jetson nano to connect to a pixhawk through USB so the endpoint should be /dev/ttyUSB0:57600? So dronekit python will tell mavsdk_server to go look for a pixhawk connected through usb?
Thanks

maxwelllwang, 2020-05-14:
`Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mavsdk/system.py", line 166, in _start_mavsdk_server
    with path(bin, 'mavsdk_server') as backend:
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/usr/lib/python3.8/importlib/resources.py", line 201, in path
    with open_binary(package, resource) as fp:
  File "/usr/lib/python3.8/importlib/resources.py", line 91, in open_binary
    return reader.open_resource(resource)
  File "<frozen importlib._bootstrap_external>", line 988, in open_resource
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/dist-packages/mavsdk/bin/mavsdk_server'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "takeoff_and_land.py", line 37, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.8/asyncio/base_events.py", line 608, in run_until_complete
    return future.result()
  File "takeoff_and_land.py", line 10, in run
    await drone.connect(system_address="udp://:14540")
  File "/usr/local/lib/python3.8/dist-packages/mavsdk/system.py", line 62, in connect
    self._start_mavsdk_server(system_address)
  File "/usr/local/lib/python3.8/dist-packages/mavsdk/system.py", line 175, in _start_mavsdk_server
    raise Exception("\nIt seems like this installation does not provide an embedded 'mavsdk_server' binary. If you installed from pip, it means that 'mavsdk_server' is not distributed for your platform (yet). You will need to get and run it manually:\n\n\t1. Build 'mavsdk_server': https://github.com/mavlink/mavsdk.\n\t2. Run it, e.g. on port 50051: './mavsdk_server -p 50051'.\n\t3. Set the 'mavsdk_server_address' and port when creating the System: 'drone = System(mavsdk_server_address='localhost', port=50051)\n")
Exception: 
It seems like this installation does not provide an embedded 'mavsdk_server' binary. If you installed from pip, it means that 'mavsdk_server' is not distributed for your platform (yet). You will need to get and run it manually:
`

Is this because I installed mavsdk incorrectly? should I try removing it using pip and trying again?

JonasVautherin, 2020-05-14:
@maxwelllwang: This issue is closed, and you are obviously not using version 0.5.2. I created a new issue for your new question: #189.