mavlink/MAVSDK-Python #156: connect(..) over Serial waiting forever..

ramgrandhi, 2020-01-25:
Hello All -

I have a connected system (Jetson Nano as CC [Companion Computer], connected to Pixhawk4's TELEM2 port via serial RX/TX/GND connections over 3.3V).

On the CC, I was able to test the Mavlink connection using **mavlink_shell.py** from PX4/Tools [here](https://github.com/PX4/Firmware/blob/master/Tools/mavlink_shell.py).

> python3 **mavlink_shell.py** /dev/ttyTHS1 --baudrate 115200
> Connecting to MAVLINK...
> NuttShell (NSH)
> nsh> 

However, when I use MavSDK-Python's examples with

JonasVautherin, 2020-01-26:
Unless you are running on macOS, Windows or Linux-x64, you need to run `mavsdk_server` manually and set `mavsdk_server_address='localhost'` when instantiating `System()`. Something like:

```python
let drone = System(mavsdk_server_address='localhost')
await drone.connect(...)
```

MAVSDK-Python should be able to detect that situation and show an instructive error message. I will look into that ASAP. In the meantime you just have to know it :sweat_drops:.

Let me know if that helps!

ramgrandhi, 2020-01-28:
Thanks @JonasVautherin.

My target platform is Ubuntu running on AARCH64, so I built mavsdk_server from source and ran it as a background process. See below.

`
[03:01:38|Info ] MAVSDK version: 0.23.0 (mavsdk_impl.cpp:25)
[03:01:38|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:401)
[03:01:38|Info ] Server started (grpc_server.cpp:38)
[03:01:38|Info ] Server set to listen on 0.0.0.0:37791 (grpc_server.cpp:39)
[03:01:38|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:22)
`

Now, I modified the example *takeoff_and_land.py* to include these lines (As per your suggestion):

`
    drone = System(mavsdk_server_address="localhost")
    await drone.connect(system_address="serial:///dev/ttyTHS1:115200")
`

Here is the output (it waits forever!):
`
Waiting for mavsdk_server to be ready...
_
`

Appreciate your thoughts!

Cheers,
ram

julianoes, 2020-01-28:
> [03:01:38|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:22)  

Your `mavsdk_server` is listening on UDP instead of serial. I think you don't need the line:

```
await drone.connect(system_address="serial:///dev/ttyTHS1:115200")   
```

But you need to start the `mavsdk_server` with the correct argument:
```
mavsdk_server serial:///dev/ttyTHS1:11520
```

JonasVautherin, 2020-01-28:
@julianoes is right, except that you still need to connect, but without the argument: `await drone.connect()`. Also, you need to be careful with the port to which `mavsdk_server` listens. By default it will choose a port and show it in the logs. But you can set it manually.

So either you run `mavsdk_server -p 50051 serial:///dev/ttyTHS1:11520`, and then in python:

```python
drone = System(mavsdk_server_address="localhost")
await drone.connect()
```

or you run `mavsdk_server serial:///dev/ttyTHS1:11520`, look at the port to which the gRPC server listens (in your logs above it was 37791) and call that in python:

```python
drone = System(mavsdk_server_address="localhost", port=37791)
await drone.connect()
```

ramgrandhi, 2020-01-28:
Yess!
It worked like magic!! Many Thanks @JonasVautherin @julianoes :)

Cheers!

closed 2020-01-28 by ramgrandhi

Safranek42, 2020-05-06:
Hi,
I have the same problem on Jetson Nano. When I tried setting up mavsdk_server with
mavsdk_server serial:///dev/ttyACM0
I got
bash: mavsdk_server: command not found.

I built MAVSDK and MAVSDK-Python from source.

JonasVautherin, 2020-05-06:
@Safranek42: this is unrelated, I created a new issue for you (#180).