mavlink/MAVSDK-Python #157: Implement goto_location

avillanueva, 2020-01-29:
It would be very useful to have goto_location command (as available in c++ API)

JonasVautherin, 2020-01-29:
It is an action, so it would need to be added to the [action interface definition here](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/action/action.proto). Then the next step would be to add it to `mavsdk_server` [here](https://github.com/mavlink/MAVSDK/blob/develop/src/backend/src/plugins/action/action_service_impl.h).

@avillanueva: Would you be willing to contribute this? :blush:

vivh3, 2020-02-27:
Is anything willing to contribute this action? Would greatly appreciate :)

coderkalyan, 2020-03-04:
@vivh3 @JonasVautherin @avillanueva I'll see what I can do, I'm trying to get more involved with implementing more features into MAVSDK-Python, but I'm a bit short on time as school work is high around this time of year, and I don't want to take on too much. I already have a couple other issues im working on for MAVSDK, I'll keep you posted!

JonasVautherin, 2020-05-13:
It is now in release 0.7.0!

closed 2020-05-13 by JonasVautherin