mavlink/MAVSDK-Python #158: Implement the follow-me plugin in Python? (ROS)

josepeck, 2020-02-14:
Hi!, is there any way (or plan) to implement the `Follow-me` plugin in Python? I want to launch a ROS node with Python that will do the following:

1) Listen to the `/rover/mavros/global_position/global` topic and extract the lat, lon coordinates of a simulated rover in Gazebo.
2) Implement the `Follow me` class in the same ROS node so it can use the obtained lat, lon coordinates and make the PX4-SITL drone follow the rover in the same simulation.

I've already compiled the C++ follow me example and it's working fine. I'm quite new to MAVSDK and C++ so any help or guide will be appreciated!

julianoes, 2020-02-14:
If you're using ROS, I would consider using MAVROS. I'm not sure if follow me is implemented in MAVROS though. I could not determine that looking at the code.

Now Follow Me should not be too hard to be added to Python because it is already added to the proto   files:
https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/followme/followme.proto

However, the C++ backend is missing here:
https://github.com/mavlink/MAVSDK/tree/develop/src/backend/src/plugins

as well as the Python part. That one, we should be able to auto-generate.

@JonasVautherin can probably helpout with the details :smile:

JonasVautherin, 2020-02-15:
Yep! @josepeck would you be willing to contribute the C++ bridge to Python? If yes I can give instructions =)

josepeck, 2020-02-18:
Hi @JonasVautherin, currently i'm a little bit short on time because of my master thesis and internship but we can take a look at it to see if i can help with something.

coderkalyan, 2020-03-04:
@josepeck @JonasVautherin I also started implementing the bridge for both Tune and FollowMe - but I've also been rather short on time due to school work and other activities. Currently, the proto files have been implemented (I actually made a couple changes, I will try to push them) but the bridging code has yet to be implemented.

If you are interested in implementing it @josepeck , I would be happy to hand it over to you, and @JonasVautherin could explain how to do it. Otherwise, I hope to find time to do it soon, in which case I will keep you updated!

PR #188 (2020-05-14): Generate new plugins: follow_me, ftp, log_files, mission_raw, tune
Seems like I had forgotten to update the plugins list for 0.7.0 :see_no_evil:.

Fixes #158.
closed 2020-05-14 by julianoes