mavlink/MAVSDK-Python #159: Stuck at  "Waiting for Drone to connect" while trying to connect drone over serial to run example takeoff-and-land.py

yash2410, 2020-02-20:
`   drone = System()
    await drone.connect(system_address="serial:///dev/ttyACM0:57600")
    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():`

The output is stuck on 
`Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
`
I checked mavlink status on QGC and gave me this output
[image]
[image]

I changed the baudrate match to QGC but it didn't make an difference

julianoes, 2020-02-20:
When you try this, is QGC shutdown so it is not blocking the serial port?

yash2410, 2020-02-24:
It was shutdown. I opened QGC just to check the mavlink status

julianoes, 2020-02-28:
What platform is this on? Can you try to run `mavsdk_server` manually to check if it is connecting?

yash2410, 2020-03-03:
This is on Ubuntu 18.04 running python3.6 and mavsdk version 0.6.1. The drone was serially connected through USB.
I have solved it now by going for a UDP connection instead.

closed 2020-03-03 by yash2410

rachoun, 2020-04-15:
Hello, can you please tell me what could be the max tx and rx rates?

JonasVautherin, 2020-04-15:
@rachoun Hello! This is off-topic here. Please open a new issue to ask your question :blush: