mavlink/MAVSDK-Python #160: How to implement area coverage?

vivh3, 2020-02-27:
Hi! I'm new to using MAVSDK-Python and am looking for suggestions for how to make the drone survey a defined polygonal area; it would start at a vertex and fly back and forth until it has flown over the entire area. Would this have to involve making a mission and populating it with many waypoints?

julianoes, 2020-02-28:
Yes, you would have to generate the waypoints/mission items for the survey that you want to fly. This is not implemented "inside" MAVSDK because it would generally functionality developed on top/using MAVSDK for a specific application, e.g. surveying, inspection, crop spraying.

There are two ways to implement the survey mission items:
1. Define all "corners" of the mission and use distance-based or time-based triggering of a camera.
2. Define each waypoint where the drone should take a picture. This will require to define and upload many more mission items but means that you have more control in specifying where pictures are taken.

vivh3, 2020-02-28:
Thanks for the info! Follow-up questions:
1. If one would rather not use missions, do you foresee a goto_location action being implemented for MAVSDK-Python?
2. Can one define a polygonal geofence such that the drone turns around when it hits an edge of the geofence?

julianoes, 2020-03-01:
1. Feel free to make a contribution to add that. First step is to add it to [MAVSDK-Proto](https://github.com/mavlink/MAVSDK-Proto) and then to the backend/mavsdk_server. The Python part will be auto-generated from proto.

2. That's not supported by PX4, and geofence is quite rudimentary for PX4 in general, so I would not recommend that.

hamishwillee, 2020-03-01:
FYI 2 may no longer be true. I THINK ardupilot is now mavlink compliant and proper geofence support present.

julianoes, 2020-03-02:
@hamishwillee in 2 I'm talking about PX4, so I'm confused what you mean.

hamishwillee, 2020-03-04:
@julianoes It means I can't read. Apologies.

JonasVautherin, 2020-05-13:
Closing for inactivity, feel free to reopen if necessary.

closed 2020-05-13 by JonasVautherin