mavlink/MAVSDK-Python #162: bus error when connecting PX4

Jabba-The-Gut, 2020-03-11:
Hello everyone,

I've setup my PX4 with a Raspberry Pi Zero W as CC (Companion Computer). I was able to setup Mavsdk-Python by manually building mavsdk from sources and cross compile the mavsdk_server.

Im manually starting the mavsdk_server binary:
```python
./mavsdk_server serial:///dev/serial0:921600
```

The server starts successfully:
```
[07:18:09|Info ] MAVSDK version: 0.24.0 (mavsdk_impl.cpp:25)
[07:18:09|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:401)
E0311 19:18:09.939134220    1230 socket_utils_common_posix.cc:208] check for SO_REUSEPORT: {"created":"@1583954289.938931222","description":"SO_REUSEPORT unavailable on compiling system","file":"/work/build/linux-armv6/third_party/grpc/grpc/src/grpc/src/core/lib/iomgr/socket_utils_common_posix.cc","file_line":176}
[07:18:09|Info ] Server started (grpc_server.cpp:38)
[07:18:09|Info ] Server set to listen on 0.0.0.0:43263 (grpc_server.cpp:39)
[07:18:09|Info ] Waiting to discover system on serial:///dev/serial0:921600... (connection_initiator.h:22)
```
As soon as I connect my PX4 using a serial connection I get "bus error":
```
[07:20:08|Debug] Component Autopilot (1) added. (system_impl.cpp:386)
[07:20:08|Debug] Discovered 1 component(s) (UUID: 3762846558525535283) (system_impl.cpp:556)
[07:20:08|Info ] System discovered [UUID: 3762846558525535283] (connection_initiator.h:61)
[07:20:09|Warn ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:222)
[07:20:09|Warn ] sending again, retries to do: 3  (CAL_GYRO0_ID). (mavlink_parameters.cpp:458)
[07:20:09|Warn ] sending again, retries to do: 2  (511). (mavlink_commands.cpp:222)
[07:20:10|Warn ] Command ack 511 not matching our current command: 520 (mavlink_commands.cpp:143)
[07:20:11|Warn ] sending again, retries to do: 3  (CAL_ACC0_ID). (mavlink_parameters.cpp:458)
Bus error
```
After Bus error the server crashes.

Now I have 2 questions:
- what does `Command ack 511 not matching our current command: 520` mean?
- what does `Bus error` mean?

Best Regards

julianoes, 2020-03-13:
> what does Command ack 511 not matching our current command: 520 mean?

This means that an ack was received for command 511 instead of 520 which it expected. This can have various reasons, I wouldn't be too worried. However, all the retries hint that maybe there is considerable packet loss.
I would try to connect with a lower baudrate first and see if that behaves ok.

> what does Bus error mean?

Interesting, I've never seen this! Since it crashes, can you run it with `gdb --args mavsdk_server serial:///dev/serial0:921600` to see if you get more information. And if it crashes you can type `backtrace` to see more as well, or `thread apply all bt`.

Jabba-The-Gut, 2020-03-22:
Thank you very much for your answer.
 
I wasn't able to collect more information using ```gdb --args mavsdk_server serial:///dev/serial0:921600```.  However, your suggestion of lowering the baudrate did the job. I still have quite some packet loss (based on your comment of the failed ack commands), but at least the connection is now stable.

I've chosen a baudrate that is suggested by QGroundControl (9100 if I remember correctly) and everything's fine.

Thank you!

closed 2020-03-22 by Jabba-The-Gut