mavlink/MAVSDK-Python #163: Param class error

siddux, 2020-03-19:
Hi, I was trying to familiarize with param class. I run this simple code:

```
async def run():

    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break

    await drone.param.set_param_int('GF_ACTION', 1)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

And the output error that it gives to me is: AttributeError: 'Param' object has no attribute 'set_param_int'. I don't know why it doesn't work. I search on generated library files and this attribute is defined on it.

closed 2020-03-19 by siddux

JonasVautherin, 2020-03-19:
Did you solve your problem? I see that your code here says `set_int_param`, and your description says `set_param_int`. Was that the issue?

siddux, 2020-03-19:
Yes, I solved it. I was looking at API reference form C++ documentation and on this page talks about set_param_int but then looking at generated files I find my mistake.  Thank you!