mavlink/MAVSDK-Python #165: How to get local ned location

canberkdurmus, 2020-03-19:
Hello everyone,

I am new with MAVSDK Python and trying to get location of the vehicle in local ned frame. I am able to get the location in global frame with `drone.telemetry.position()` but I need to convert it into local frame in every target point reach check. I thought, since we are able to set target location with local ned frame, there must be the get part but I can not find it in the examples nor in the documentation.

I am looking for the MAVSDK Python equivalent of the `vehicle.location.global_relative_frame` from Dronekit. What is the best way to do it?

JonasVautherin, 2020-03-19:
@irsdkv can you help with that?

julianoes, 2020-03-20:
I think that's currently not available in Python yet but only in C++:
https://github.com/mavlink/MAVSDK/blob/0b305884164c3e7f3bce98f29ab3b1df13d42557/src/plugins/telemetry/include/plugins/telemetry/telemetry.h#L925-L930

canberkdurmus, 2020-03-20:
Thank you so much for your response guys, we will try to add it or convert global lat lon to local x y related to the home position.

julianoes, 2020-03-20:
@JonasVautherin you had helper functions for global local transformations in Python somewhere, right?

canberkdurmus, 2020-03-20:
That was helpful, for now I am going to implement the python version of the `local_from_global` function for our python base.

As far as I can see that MAVSDK Python is connecting to the MAVSDK Server with gRPC and MAVSDK Server is getting required information from MAVSDK Cpp. I planning to add it to the MAVSDK Server in our own fork.

Thank you so much again 👍

julianoes, 2020-03-20:
> I planning to add it to the MAVSDK Server in our own fork.

Why is that required?

When you do `pip install mavsdk` the `mavsdk_server` is included, however, that only works for Win, macOS and Linux x86_64 but not for arm architectures like Raspberry Pi yet.

canberkdurmus, 2020-03-20:
I think MAVSDK Server should have something like an endpoint to get the local ned position information from the MAVSDK Cpp to send back to the MAVSDK Python. So I think, to make it available in python we will need to make changes on backend too. I will make a deeper research about MAVSDK Server later.

To use it in arm based systems I am planning to build it from source with `make BUILD_BACKEND=YES`

JonasVautherin, 2020-03-20:
Here it is in Python: 

```python
def global_from_local(north_m, east_m, ref_lat, ref_lon):
    world_radius_m = 6371000
    ref_lat_rad = math.pi / 180 * ref_lat
    ref_lon_rad = math.pi / 180 * ref_lon
    x_rad = north_m / world_radius_m
    y_rad = east_m / world_radius_m

    c = math.sqrt(x_rad * x_rad + y_rad * y_rad);

    if (c == 0):
        return ref_lat, ref_lon

    sin_c = math.sin(c)
    cos_c = math.cos(c)

    ref_sin_lat = math.sin(ref_lat_rad)
    ref_cos_lat = math.cos(ref_lat_rad)

    lat_rad = math.asin(cos_c * ref_sin_lat + (x_rad * sin_c * ref_cos_lat) / c)
    lon_rad = (ref_lon_rad + math.atan2(y_rad * sin_c, c * ref_cos_lat * cos_c - x_rad * ref_sin_lat * sin_c))

    global_lat = 180 / math.pi * lat_rad
    global_lon = 180 / math.pi * lon_rad

    return global_lat, global_lon
```

canberkdurmus, 2020-03-20:
I was just finished implementation and testing according to the cpp version. Will compare it with this one also, thank you so much @JonasVautherin 👍

closed 2020-03-20 by canberkdurmus