mavlink/MAVSDK-Python #167: How to connect serial in Windows?

LiangR2017, 2020-03-24:
Hope u still fine.and of course hello to u.
Find your codes
Serial: 

> serial:///path/to/serial/dev[:baudrate]

but it seems to work on linux only
Have tried   `awaitdrone.connect(system_address="serial://com14:115200")` but didn't work. 
Can u help me?thanks

LiangR2017, 2020-03-25:
also tried 

> await drone.connect(system_address="serial:com14:115200")

doesnt work .

LiangR2017, 2020-03-25:
trying to install unbuntu now .still want to know how to connect in windows
tell me if anybody knows.thanks a lot!

julianoes, 2020-03-25:
@LiangR2017 what error do you get when you try that?

Can you try `serial://COM14:115200`

LiangR2017, 2020-03-25:
@julianoes just show these and i also tried " serial://COM14:115200"
> Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...

maxwelllwang, 2020-05-15:
what are the parameters to connect with mavsdk_server manually? when you run ./mavsdk_server i know you can specify port as -p 50051. how do you specify connection address?

maxwelllwang, 2020-05-15:
Ok I successfully connected on ubuntu with ./mavsdk_server -p 50051 serial///dev/ttyACM0:57600 
but its telling me im sending invalid commands

JonasVautherin, 2020-05-15:
I'm assuming you meant `./mavsdk_server -p 50051 serial:///dev/ttyACM0:57600` (note the `:`).

> its telling me im sending invalid commands

What are you sending, and what does it say precisely? :sweat_smile:

maxwelllwang, 2020-05-15:
I think you addressed it in the other thread because my pixhawk is running ardupilot not pixhawk
Oops forgot the `:`

closed 2020-05-15 by julianoes