mavlink/MAVSDK-Python #168: Take off only works once

Emrage, 2020-03-24:
Hello,

Whenever I try to take off more than once it doesn't work.

`import asyncio
from abc import ABC

import logger
from drone.drone_action import DroneAction
from drone.drone_state import DroneState

class TakeOffAction(DroneAction, ABC):
    log = logger.init_logger(__name__, "info")

    async def action(self, drone_manager):
        self.log.info(f'Starting take_off action, state={drone_manager.drone_info.state}, args={self.args} '
                      f'info={drone_manager.drone_info.to_string()}')

        if not drone_manager.drone_info.in_air:

            if not drone_manager.drone_info.armed:
                arm_action = drone_manager.actions['arm']
                arm_action.supply(arm_action, args=None)
                await arm_action.action(arm_action, drone_manager)

            drone_manager.drone_info.state = DroneState.TAKING_OFF

            await drone_manager.drone.action.takeoff()

            await asyncio.sleep(5)

            drone_manager.drone_info.taken_off = True

            self.log.info(f'Finished take_off action, state={drone_manager.drone_info.state}, args={self.args} '
                          f'info={drone_manager.drone_info.to_string()}')
`

This is the code for taking off. But landing does work multiple times when I just type 

> commander takeoff

In the console and then call this function again.

This is the land code

`    log = logger.init_logger(__name__, "info")

    async def action(self, drone_manager):
        self.log.info(f'Starting land_action action, state={drone_manager.drone_info.state}, args={self.args} '
                      f'info={drone_manager.drone_info.to_string()}')

        if drone_manager.drone_info.in_air:

            if not drone_manager.drone_info.armed:
                arm_action = drone_manager.actions['arm']
                arm_action.supply(arm_action, args=None)
                await arm_action.action(arm_action, drone_manager)

            drone_manager.drone_info.state = DroneState.LANDING

            await drone_manager.drone.action.land()

            await asyncio.sleep(5)
            drone_manager.drone_info.taken_off = False

            self.log.info(f'Finished land_action action, state={drone_manager.drone_info.state}, args={self.args} '
                          f'info={drone_manager.drone_info.to_string()}')`

And this is the code for the arm action

`class ArmAction(DroneAction, ABC):
    log = logger.init_logger(__name__, "info")

    async def action(self, drone_manager):
        self.log.info(f'Starting arm action, state={drone_manager.drone_info.state}, args={self.args} '
                      f'info={drone_manager.drone_info.to_string()}')

        if not drone_manager.drone_info.armed:
            drone_manager.drone_info.state = DroneState.ARMING
            await drone_manager.drone.action.arm()
            drone_manager.drone_info.armed = True

        self.log.info(f'Finished arm action, state={drone_manager.drone_info.state}, args={self.args} '
                      f'info={drone_manager.drone_info.to_string()}')
`

Everything gets called correctly and I also get no exceptions.

Emrage, 2020-03-24:
Figured it out, when landing the drone also disarms

closed 2020-03-24 by Emrage

LiangR2017, 2020-03-25:
hello,  only find u commentd recently . want to ask some question
i tried to run the land and take off codes on my windows through my usb serial
but i found the only example
  

> Serial:` serial:///path/to/serial/dev[:baudrate]

what's its like on windows?i tried the follow but it dosent work,

> await drone.connect(system_address="serial:COM13:115200")

can you help me?

Emrage, 2020-03-25:
> hello, only find u commentd recently . want to ask some question
> i tried to run the land and take off codes on my windows through my usb serial
> but i found the only example
>
>
> what's its like on [...]

Hello, I'm not sure how it would work with the serial but I have found this:

https://github.com/mavlink/MAVSDK-Python/issues/100

and this

https://discuss.px4.io/t/serial-instead-of-udp-for-dronecore-examples/5459/3

The people there are using 

> serial:///dev/ttyACM0

and 

> serial:///dev/ttyUSB0:57600

So from what I think yours needs to be

serial:///dev/ttyUSB13:115200

or 

serial:///dev/ttyCOM13:115200

JonasVautherin, 2020-03-26:
@LiangR2017: It seems to me that your issue is about connecting over a serial link and has nothing to do with the original issue here. Is that correct? If yes, would you mind opening a new issue with your question? :blush: