mavlink/MAVSDK-Python #170: Monitor QGC mission progress

erenaud3, 2020-04-02:
Hi !

I am trying to monitor QGC mission progress. I have the following code (greatly inspired from example/mission.py):

```python
#!/usr/bin/env python3

import asyncio

from mavsdk import System
from mavsdk import (MissionItem)

async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    asyncio.ensure_future(print_mission_progress(drone))

async def print_mission_progress(drone):
    async for mission_progress in drone.mission.mission_progress():
        print(f"Mission progress: "
              f"{mission_progress.current_item_index}/"
              f"{mission_progress.mission_count}")

if __name__ == "__main__":
    # Start the main function
    asyncio.ensure_future(run())

    # Runs the event loop until the program is canceled with e.g. CTRL-C
    asyncio.get_event_loop().run_forever()
```

Here how I proceed:
- start JMavSim HITL simulation
- start Python script above
- start QGC and upload dummy mission

I was expecting to see something like:  "Mission progress: 0/{MAX}" right after upload and then see a new line for each waypoint.

I only get "Mission progress: -1/0" when I start script and nothing after mission upload nor during mission.

Am I missing something ?

julianoes, 2020-04-03:
Is HITL simulation actually working with fmu-v2? I don't expect HITL to work because `pwm_out_sim` is missing: https://github.com/PX4/Firmware/blob/ff1e5a595dc87997e063308b8c8be0f539994a1c/boards/px4/fmu-v2/default.cmake#L57.

Is the mission running in QGC?

One thing to note for mission progress of the `mission` plugin is that it only works properly for missions uploaded using MAVSDK. For other missions I would recommend to use the `mission_raw` plugin, however, that one is not exposed in Python yet (but it's coming soon!).

erenaud3, 2020-04-03:
> I don't expect HITL to work because pwm_out_sim is missing: https://github.com/PX4/Firmware/blob/ff1e5a595dc87997e063308b8c8be0f539994a1c/boards/px4/fmu-v2/default.cmake#L57.

First of all, thank you very much for that. I didn't realize `pwm_out_sim` was vital for HITL until you pointed it out !

> Is HITL simulation actually working with fmu-v2?
> Is the mission running in QGC?

Actually, I got really lucky. Because I tried to work with branch release/v1.10, on which `pwm_out_sim` is NOT commented (https://github.com/PX4/Firmware/blob/release/v1.10/boards/px4/fmu-v2/default.cmake).
So yes, in that case the mission running in QGC and the drone flies accordingly in JMavSim.

On `master` and `release/1.10` branches, HITL wasn't working obviously. But oddly enough, it works when I upload latest stable fmu-v2 release from QGC. 

> One thing to note for mission progress of the mission plugin is that it only works properly for missions uploaded using MAVSDK. 

I was expecting something like that. 

> For other missions I would recommend to use the mission_raw plugin, however, that one is not exposed in Python yet (but it's coming soon!).

That's great news ! Thank you for your answer :+1:

closed 2020-04-03 by erenaud3