mavlink/MAVSDK-Python #172: Multiple simulation

siddux, 2020-04-06:
I have been trying to do multiple simulation using jmavsim but I think I get some problems with uuid because if I simulate 2 drones both get the same uuid and this makes the program to crash. I simply test this using de takeoffandland example and duplicating the commands to address drone 1, with udp port 14540 and drone 2 with udp port 14541. 

I have seen some issues in C library forum but I can't get it working. The simulation environment is working properly because I can fly both drones addressing to first one and, when it has landed, addressing to the other changing the udp port. 

Any tip about how to solve it? Is multithreading the solution?

JonasVautherin, 2020-04-06:
If they use a different MAVLink interface (here you mention 14540 and 14541), then the uuid or sysid don't matter, it should just work. The only thing in MAVSDK-Python is that you need to start `mavsdk_server` manually.

Can you give more details about what you tried and how it crashes? Some code, for instance :blush:

siddux, 2020-04-06:
First I run simulator using instructions on px4 dev guide. The code I run is this:

`#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone1 = System()
    drone2 = System()
    await drone1.connect(system_address="udp://:14540")
    await drone2.connect(system_address="udp://:14541")

    print("Waiting for drone to connect...")
    async for state in drone1.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Waiting for drone to connect...")
    async for state in drone2.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone1.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone2.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break

    print("-- Arming")
    await drone1.action.arm()
    await drone2.action.arm()

    print("-- Taking off")
    await drone1.action.takeoff()
    await drone2.action.takeoff()

    await asyncio.sleep(5)

    print("-- Landing")
    await drone1.action.land()
    await drone2.action.land()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
`
It's not an elegant code but is only for test and understand multisim interaction with mavsdk. Executing this, first drones takeoff and land but before the second one can take of it crashes with the following error message:

`Traceback (most recent call last):
  File "multi_sim.py", line 54, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "multi_sim.py", line 50, in run
    await drone2.action.land()
  File "/home/edu/.local/lib/python3.6/site-packages/mavsdk/generated/action.py", line 300, in land
    raise ActionError(result, "land()")
mavsdk.generated.action.ActionError: COMMAND_DENIED: 'Command denied'; origin: land(); params: ()
`

JonasVautherin, 2020-04-07:
You need to run two instances of `mavsdk_server` manually. So something like:

```
./mavsdk_server -p 50050
```

would go together with:

```
drone1 = System(mavsdk_server_address="127.0.0.1", port=50050)
```

and e.g. 

```
./mavsdk_server -p 50051
```

would go with

```
drone1 = System(mavsdk_server_address="127.0.0.1", port=50051)
```

siddux, 2020-04-07:
This is exactly what I needed. Thank you very much for your help!

closed 2020-04-07 by siddux

gonmarmar5, 2023-01-23:
Hi! I've tried to do the same multiple connection but it throws me the same error showed here. I've tried to run the `./mavsdk_server -p 50050` but i don't have the file mavsdk_server. I've read in others threads (https://github.com/mavlink/MAVSDK-Python/issues/180) that it can be manually installed, is there anyone that could help me install this file? I'm working on python 3.8. Thanks!

siddux, 2023-01-23:
You have to[ build the library with MAVSDK SERVER flag](https://mavsdk.mavlink.io/main/en/cpp/guide/build_mavsdk_server.html):

`cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF -DBUILD_MAVSDK_SERVER=ON -Bbuild \
cmake --build build/ -j8`

It's a little bit tricky to find the executable after the build as it is built in `MAVSDK/build/src/mavsdk_server/src/`