mavlink/MAVSDK-Python #174: Local positioning not exposed in python yet

sudo-raza, 2020-04-14:
Hello 

I am just bringing up this issue here again for reference as Julian Oes is already aware and hard at work. 

Local positioning (positionNED) on MAVSDK-python is not available. Therefore you are unable to get local positioning for the drone. However global coordaintes (position in the telemetry class) are available. 

Simulation: jMAVSim
OS: Windows 10

julianoes, 2020-04-15:
Thanks for the note @sudo-raza.

This is coming with: https://github.com/mavlink/MAVSDK-Python/pull/171/files#diff-2d7f0ba673b8f8a848712bbff51092e0R1549

MSBGit, 2020-04-25:
I built the MAVSDK develop branch and the MAVSDK-Python branch that @julianoes pointed me to in the PX4 #mavsdk Slack channel. In general I would say this is a tremendous improvement in functionality. What I really wanted is to get NED Position and Velocity updates and to be able to set telemetry rates.

There are a few things that seem broken and I'm not sure if it's because of my environment or because the code needs to be fixed.

What I am seeing is that a lot of the APIs I'm using are now throwing exceptions even when the actual API call into the Pixhawk succeeds. I can arm, takeoff, fly using Offboard NED coordinates and then land and disarm, but I get exceptions for every call. Ignoring he exceptions (as in my example below) allow the drone to fly the way I want it to. It doesn't seem that an API call that succeeds should throw an exception, but I wonder if that's the new model for this update?

I am also now receiving NED Position and Velocity updates (thanks!) but haven't verified if the values I am getting back are correct.

```
-- Arming
Ignoring ActionError while arming drone: <class 'mavsdk.generated.action.ActionError'> None: 'Success: the action command was accepted by the vehicle'; origin: arm(); params: ()

-- Setting initial setpoint
Ignoring exception while set_position_ned: <class 'mavsdk.generated.offboard.OffboardError'> None: 'Request succeeded'; origin: set_position_ned(); params: (<mavsdk.generated.offboard.PositionNedYaw object at 0x7fb628e358>,)

-- Starting offboard
Ignoring starting offboard mode exception: <class 'mavsdk.generated.offboard.OffboardError'> None: 'Request succeeded'; origin: start(); params: ()

-- Go 0m North, 0m East, -100m Down within local coordinate system
Ignoring exception: <class 'mavsdk.generated.offboard.OffboardError'> None: 'Request succeeded'; origin: set_position_ned(); params: (<mavsdk.generated.offboard.PositionNedYaw object at 0x7fb5218780>,)

-- Go 100m North, 0m East, -100m Down within local coordinate system, turn to face East
Ignoring exception: <class 'mavsdk.generated.offboard.OffboardError'> None: 'Request succeeded'; origin: set_position_ned(); params: (<mavsdk.generated.offboard.PositionNedYaw object at 0x7fb628e358>,)

-- Go 5m North, 200m East, -100m Down within local coordinate system, turn 90degrees right
Ignoring exception: <class 'mavsdk.generated.offboard.OffboardError'> None: 'Request succeeded'; origin: set_position_ned(); params: (<mavsdk.generated.offboard.PositionNedYaw object at 0x7fb5218780>,)

-- Go 400m North, 10m East, -100m Down within local coordinate system, turn to face South
Ignoring exception: <class 'mavsdk.generated.offboard.OffboardError'> None: 'Request succeeded'; origin: set_position_ned(); params: (<mavsdk.generated.offboard.PositionNedYaw object at 0x7fb628e358>,)

-- Stopping offboard
Stopping offboard mode failed with error code: <class 'mavsdk.generated.offboard.OffboardError'> None: 'Request succeeded'; origin: stop(); params: ()

-- Landing
Ignoring exception: <class 'mavsdk.generated.action.ActionError'> None: 'Success: the action command was accepted by the vehicle'; origin: land(); params: ()

-- Waiting for drone to land
-- Drone has landed

-- Disarming
Ignoring exception: <class 'mavsdk.generated.action.ActionError'> None: 'Success: the action command was accepted by the vehicle'; origin: disarm(); params: ()
```

JonasVautherin, 2020-04-25:
> What I am seeing is that a lot of the APIs I'm using are now throwing exceptions even when the actual API call into the Pixhawk succeeds.

This to me is a hint that you are not using matching versions of `mavsdk_server` and `MAVSDK-Python`. With the auto-generation, I think I changed the order in the `Offboard::Result` enum, meaning that a new `mavsdk_server` build will send `SUCCESS` through protobuf, but an older MAVSDK-Python will read an error (because it's just a number over protobuf, and `SUCCESS` changed from "0" to "1").

This won't happen anymore once the API is stable (when we move to mavsdk 1.0), but right now that's an effect of those changes, sorry :see_no_evil:.

Which MAVSDK-Python branch are you using, and are you sure it is up-to-date?

MSBGit, 2020-04-26:
> > What I am seeing is that a lot of the APIs I'm using are now throwing exceptions even when the actual API call into the Pixhawk succeeds.
> 
> This to me is a hint that you are not using matching versions of `mavsdk_server` and `MAVSDK-Python`. With the auto-generation, I think I changed the order in the `Offboard::Result` enum, meaning that a new `mavsdk_server` build will send `SUCCESS` through protobuf, but an older MAVSDK-Python will read an error (because it's just a number over protobuf, and `SUCCESS` changed from "0" to "1").
> 
> This won't happen anymore once the API is stable (when we move to mavsdk 1.0), but right now that's an effect of those changes, sorry 🙈.
> 
> Which MAVSDK-Python branch are you using, and are you sure it is up-to-date?

Here is what I'm building with:

MSBGit, 2020-04-26:
I rebuilt `MAVSDK-Python` on my nano (which was very painful because of the long time it took) and now my test program works without having the exceptions thrown from `MAVSDK`:

```
$ python3 moveNEDmavsdk.py
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
MAVSDK server connected at: localhost
Drone discovered with UUID: 3546673793709193016
Waiting for drone to have a global position estimate...
health = Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_level_calibration_ok: True, is_local_position_ok: True, is_global_position_ok: True, is_home_position_ok: True]
Global position estimate ok
-- Arming
-- Setting initial setpoint
-- Starting offboard
-- Go 0m North, 0m East, -100m Down within local coordinate system
-- Go 100m North, 0m East, -100m Down within local coordinate system, turn to face East
-- Go 5m North, 200m East, -100m Down within local coordinate system, turn 90degrees right
-- Go 400m North, 10m East, -100m Down within local coordinate system, turn to face South
-- Stopping offboard
-- Landing
-- Waiting for drone to land
-- Drone has landed
-- Disarming
```

`mavsdk_server` output (I didn't rebuild `mavsdk_server` today, I used the binary that I built yesterday):

```
$ ./mavsdk_server serial:///dev/ttyTHS1:115200 -p 50051
[11:24:43|Info ] MAVSDK version: 0.24.0-213-g1c1aae62 (mavsdk_impl.cpp:26)
[11:24:43|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:379)
[11:24:43|Info ] Server started (grpc_server.cpp:38)
[11:24:43|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:39)
[11:24:43|Info ] Waiting to discover system on serial:///dev/ttyTHS1:115200... (connection_initiator.h:22)
[11:24:43|Debug] Component Autopilot (1) added. (system_impl.cpp:352)
[11:24:43|Debug] New: System ID: 51 Comp ID: 68 (mavsdk_impl.cpp:379)
[11:24:43|Debug] Component Unsupported component (68) added. (system_impl.cpp:352)
[11:24:44|Debug] Discovered 1 component(s) (UUID: 3546673793709193016) (system_impl.cpp:525)
[11:24:44|Info ] System discovered [UUID: 3546673793709193016] (connection_initiator.h:61)
[11:24:52|Debug] MAVLink: critical: Failsafe enabled: no RC (system_impl.cpp:257)
[11:24:53|Debug] MAVLink: info: Failsafe mode activated (system_impl.cpp:257)
[11:24:53|Debug] MAVLink: info: [logger] file: /fs/microsd/log/2020-04-26/06_24_5 (system_impl.cpp:257)
[11:24:54|Debug] MAVLink: info: Failsafe mode deactivated (system_impl.cpp:257)
[11:24:54|Debug] MAVLink: info: RTL HOME activated (system_impl.cpp:257)
[11:24:54|Debug] MAVLink: info: Takeoff detected (system_impl.cpp:257)
[11:25:13|Warn ] RTT too high for timesync: 11.127 ms. (timesync.cpp:90)
[11:26:03|Warn ] RTT too high for timesync: 20.615 ms. (timesync.cpp:90)
[11:26:32|Debug] MAVLink: critical: Failsafe enabled: no RC (system_impl.cpp:257)
[11:26:33|Debug] MAVLink: info: Failsafe mode activated (system_impl.cpp:257)
[11:26:33|Debug] MAVLink: info: Failsafe mode deactivated (system_impl.cpp:257)
[11:26:34|Debug] MAVLink: info: Landing at current position (system_impl.cpp:257)
[11:27:13|Warn ] RTT too high for timesync: 15.437 ms. (timesync.cpp:90)
[11:27:58|Warn ] RTT too high for timesync: 86.767 ms. (timesync.cpp:90)
[11:28:21|Debug] MAVLink: info: Landing detected (system_impl.cpp:257)
[11:28:23|Debug] MAVLink: info: DISARMED by Auto disarm initiated (system_impl.cpp:257)
[11:28:28|Warn ] RTT too high for timesync: 16.819 ms. (timesync.cpp:90)
```

JonasVautherin, 2020-04-26:
> I rebuilt MAVSDK-Python on my nano (which was very painful because of the long time it took)

I assume you wanted to say MAVSDK, because there is nothing to build for MAVSDK-Python (just install a few things). Yes, building `mavsdk_server` takes time, which is why we try to distribute it as much as we can :innocent:.

> and now my test program works without having the exceptions thrown from MAVSDK

So it seems like it is working, right?

MSBGit, 2020-04-26:
> EDIT: oh, maybe you actually had to build gRPC itself, because it is not distributed for your particular system. Could it be?

Yes I think that was the issue.

> > and now my test program works without having the exceptions thrown from MAVSDK
> 
> So it seems like it is working, right?

Yes it is, thanks.

When will this functionality be released?

JonasVautherin, 2020-04-29:
We are finishing auto-generating the plugins in C++, I would say in less than 2 weeks :crossed_fingers:.

closed 2020-04-29 by JonasVautherin