mavlink/MAVSDK-Python #175: Can't run mission.py

Mark718real, 2020-04-26:
Hello everyone, I am new at this. I want to test example scripts. So I can not run the mission.py file which is in "example" folder. All other files work well. Here is an error.

> $ python mission.py
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 5283920058631409231
-- Uploading mission
Mission progress: 0/3
-- Arming
Mission progress: 0/3
-- Starting mission
Traceback (most recent call last):
  File "mission.py", line 92, in <module>
    loop.run_until_complete(run())
  File "C:\Users\Aziz\AppData\Local\Programs\Python\Python38-32\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "mission.py", line 63, in run
    await drone.mission.start_mission()
  File "C:\Users\Aziz\AppData\Local\Programs\Python\Python38-32\lib\site-packages\mavsdk\generated\mission.py", line 672, in start_mission
    raise MissionError(result, "start_mission()")
mavsdk.generated.mission.MissionError: ERROR: 'Error'; origin: start_mission(); params: ()

JonasVautherin, 2020-04-27:
Hello! How do you install MAVSDK? Using pip?

Does it work to takeoff from mavsdk?

Mark718real, 2020-04-27:
Yes, i installed MAVSDK. I used this guide https://auterion.com/getting-started-with-mavsdk-python/. Takeoff from mavsdk works. All other .py files like telemetry, offboard_attitude work, I can run them. But only mission.py can't start.

JonasVautherin, 2020-04-27:
Can you try to connect QGC, run `mission.py`, and then try to start the mission from QGC? You could also download it there. Just to try to see why the drone refuses to start the mission...

Mark718real, 2020-04-27:
I tried and it works). Sorry maybe for the  "stupid" question, but is it necessary to connect QGC when you run the mission.py?

JonasVautherin, 2020-04-28:
No, it's not necessary. It's just a convenient way to have a "second opinion" about the state of the drone, so to say :laughing:. When I'm working with MAVSDK and e.g. try to upload a mission, I like to download it back in QGC and inspect it there, for instance.

> I tried and it works

What worked exactly? So you started QGC, and then `mission.py` worked? It should not be related (unless PX4 is in a weird state and requires QGC). What simulator are you using?

Mark718real, 2020-04-28:
Yesterday, it did not work until I started the QGC. Without starting the QGC, it showed me the error I wrote above. I started QGC and then it worked. But now, for reasons I don't understand, it works. I am using windows 10 and JMAVSIM simulator, python 3.8.2.

JonasVautherin, 2020-04-28:
Ok, seems like it's fixed then :laughing:. Feel free to reopen if the error comes back.

closed 2020-04-28 by JonasVautherin