mavlink/MAVSDK-Python #179: goto_location function

KatrijnChys, 2020-05-05:
Hello

I'm in my third bachelor engineering technology. For my bachelorsassignment we have to write code for an autonomous flight test.
We're supposed to write a mission using some specific functions. One of the functions we miss is a 'goto_location' function.
Right now I'm trying to convert the distance in degree's between two coördinates into a distance in meters but I still get some errors.
I am using following code:
`
import math
"""radius of the Earth"""
r = 6373.0
class Waypoint():
    def __init__(self, lat, lon):
        self.lat = lat
        self.lon = lon

    async def distancelat(self, wp1, wp2):
        """LATITUDE"""
        self.wp1 = wp1
        self.wp2 = wp2
        lat1rad = math.radians(wp1.lat)
        lat2rad = math.radians(wp2.lat)

        # change in coordinates
        dlon = 0
        dlat = lat2rad - lat1rad
        # Haversine formula
        a1 = math.sin(dlat / 2) ** 2 + math.cos(lat1rad) * math.cos(lat2rad) * math.sin(dlon / 2) ** 2
        c1 = 2 * math.atan2(math.sqrt(a1), math.sqrt(1 - a1))
        distancelat = float(r * c1*1000)
        return distancelat

    async def distancelon(self, wp1, wp2):
        """LONGITUDE"""
        self.wp1 = wp1
        self.wp2 = wp2
        lon1rad = math.radians(wp1.lon)
        lon2rad = math.radians(wp2.lon)
        lat1rad = math.radians(wp1.lat)
        lat2rad = math.radians(wp2.lat)

        # change in coordinates
        dlon = lon2rad - lon1rad
        dlat = 0
        # Haversine formula
        a = math.sin(dlat / 2) ** 2 + math.cos(lat1rad) * math.cos(lat2rad) * math.sin(dlon / 2) ** 2
        c = 2 * math.atan2(math.sqrt(a), math.sqrt(1 - a))
        distancelong = float(r * c*1000)
        return distancelong
`

By using using the offboard function 'set_position_ned' I try to make the drone fly to it's new location.

`wp1 = Waypoint(47.3977395, 8.5455929)
wp2 = Waypoint(47.398039859999997, 8.5455725400000002)
mnorth = Waypoint.distancelat(wp1, wp2)
meast = Waypoint.distancelon(wp1, wp2)
await drone.offboard.set_position_ned(
    (PositionNedYaw(mnorth, meast, -5, 0)))
`

When trying to complete the mission I get the errors 'no RC and no offboard'. Another error I get is that parameter 'wp2' is ''unfilled''.

Is there any way i can solve this? Is there an easier option?

Hope to hear from you soon and thanks in advance
Katrijn Chys

JonasVautherin, 2020-05-05:
`goto_location` will come in the next release, actually :-). And it should come very, very soon. Would that help?

KatrijnChys, 2020-05-05:
Hello

Thank you for the quick answer.

The final presentation of our bachelors assignment is next Thursday but due to circumstances we were only able to start programming two weeks ago. I don't know if it will be in time for us, but it surely will be on time for the next group.

Is there  a way to request the coördinates of the launch position?

I also have a few question about the script 'mission.py'. I wasn't able to make the drone loiter above a waypoint. I entered this in the code:

mission_items.append(MissionItem(47.398039859999997,
                                 8.5455725400000002,
                                 3,
                                 10,
                                 False,
                                 float('nan'),
                                 float('nan'),
                                 MissionItem.CameraAction.NONE,
                                 float(30),
                                 float('nan')))

When running the script I get this error.

[cid:8f9b94f7-388f-4b5e-a832-160c6c28bc9f]

What do I have to enter to make the drone loiter for 30 seconds above waypoint 2?

We have to make the drone fly from launch to waypoint 1, to waypoint 2, back to launch twice in a row without landing in between and with different altitudes. Is there a way to make the drone return to launch and not land?

Thanks in advance!

Katrijn Chys

KatrijnChys, 2020-05-05:
The error I get while running the script is ''Distance between waypoints too close: 0 meters''

julianoes, 2020-05-06:
Ok, got it. That's something to look into for me.

JonasVautherin, 2020-05-06:
I have seen the "Distance between waypoints too close: 0 meters" issue recently (when working with MAVSDK-Java). The issue I had there was that my `mavsdk_server` was not up-to-date (I was using the latest protos for MAVSDK-Java).

@KatrijnChys: how do you install MAVSDK-Python? Using `pip install`?

KatrijnChys, 2020-05-06:
To install it I opended settings in PyCharm and clicked the '+' under 'project interpreter'. Searched for 'mavsdk' and clicked on  'install package'. 

This is how my teacher taught us.

JonasVautherin, 2020-05-06:
I guess PyCharm is using `pip` under the hood. So that's fine.

Does `mission.py` work if you don't modify it at all?

KatrijnChys, 2020-05-06:
When I run it without modifying it works perfectly. Also when adjusting the coördinates and altitude it workt great.

JonasVautherin, 2020-05-06:
Right. So maybe @julianoes is right and MAVSDK is doing something weird with the `loiter_time` (I would guess something like adding an extra waypoint, which is refused by PX4).

JonasVautherin, 2020-05-13:
MAVSDK-Python 0.7.0 is out, and contains `goto_location(...)`!

Regarding the `loiter_time`, @KatrijnChys would you mind trying again with the latest release and open a new issue if that's still failing?

closed 2020-05-13 by JonasVautherin

KatrijnChys, 2020-05-14:
?I'm currently having some troubles with connecting the drone to the code. I'm not able to run any scripts form examples. For example when running the mission.py script, I don't get any error's. I only get "Waiting for mavsdk_server to be ready...".

I tried reïnstalling the simulator and the mavsdk library but it still doesn't work.

julianoes, 2020-05-14:
@KatrijnChys how are you starting `mavsdk_server`?

You now need to add the gRPC port `mavsdk_server -p 50051`.

julianoes, 2020-05-14:
@JonasVautherin @KatrijnChys 

> The error I get while running the script is ''Distance between waypoints too close: 0 meters''

I can reproduce that problem. I've created an issue: https://github.com/PX4/Firmware/issues/14904

KatrijnChys, 2020-06-23:
> @KatrijnChys how are you starting `mavsdk_server`?
> 
> You now need to add the gRPC port `mavsdk_server -p 50051`.

How do I add the gPRC port?

JonasVautherin, 2020-06-23:
This is not related to `goto_location`, is it? Can  you open a new issue, explain your setup, and show us the code that opens the connection? We will answer there.

AzeemMuzammil, 2022-01-19:
@julianoes In MAVSDK_Python I am using the goto_location function and it takes lat, lon, alt and yaw as params. So, I what I want is, I have to move the drone let's say 20ms north and 15ms west from current location, how to do that. Currently I am using a function to convert lat, lon into north and east, but it's not working as expected.

```
def get_location_offset_meters(original_location, dNorth, dEast, alt):
    earth_radius = 6378137.0 #Radius of "spherical" earth
    #Coordinate offsets in radians
    dLat = dNorth / earth_radius
    dLon = dEast / (earth_radius *math.cos(math.pi * original_location.latitude_deg / 180))

    #New position in decimal degrees
    new_lat = original_location.latitude_deg + (dLat * 180 / math.pi)
    new_lon = original_location.longitude_deg + (dLon * 180 / math.pi)
    new_alt = original_location.absolute_altitude_m + alt
    new_relative_alt = original_location.relative_altitude_m + alt
    return Position(new_lat, new_lon, new_alt, new_relative_alt)
```

could you please look into it?

julianoes, 2022-01-20:
@JonasVautherin didn't you write the code for these conversions somewhere at some point?

JonasVautherin, 2022-01-20:
You mean this?

```
def global_from_local(north_m, east_m, ref_lat, ref_lon):
    world_radius_m = 6371000
    ref_lat_rad = math.pi / 180 * ref_lat
    ref_lon_rad = math.pi / 180 * ref_lon
    x_rad = north_m / world_radius_m
    y_rad = east_m / world_radius_m
​
    c = math.sqrt(x_rad * x_rad + y_rad * y_rad);
​
    if (c == 0):
        return ref_lat, ref_lon
​
    sin_c = math.sin(c)
    cos_c = math.cos(c)
​
    ref_sin_lat = math.sin(ref_lat_rad)
    ref_cos_lat = math.cos(ref_lat_rad)
​
    lat_rad = math.asin(cos_c * ref_sin_lat + (x_rad * sin_c * ref_cos_lat) / c)
    lon_rad = (ref_lon_rad + math.atan2(y_rad * sin_c, c * ref_cos_lat * cos_c - x_rad * ref_sin_lat * sin_c))
​
    global_lat = 180 / math.pi * lat_rad
    global_lon = 180 / math.pi * lon_rad
​
    return global_lat, global_lon
```

julianoes, 2022-01-20:
Yes :smile:

MuhammadBilal1, 2023-04-06:
> 

Hi, @julianoes Do we have this feature now in MavSDK Python?

I read the documentation and it says about "[set_current_mission_item](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/mission.html#mavsdk.mission.Mission.set_current_mission_item)" method. But that method is only getting the index of the current mission, which should I do if I want to give the coordinates of the new location?

julianoes, 2023-04-10:
@MuhammadBilal1 yes: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/action.html#mavsdk.action.Action.goto_location