mavlink/MAVSDK-Python #180: mavsdk_server: command not found

JonasVautherin, 2020-05-06:
I am trying to run MAVSDK on a Jetson Nano. When I tried setting up mavsdk_server with

```
mavsdk_server serial:///dev/ttyACM0
```

I got

```
bash: mavsdk_server: command not found.
```

I built MAVSDK and MAVSDK-Python from source.

_Originally posted by @Safranek42 in https://github.com/mavlink/MAVSDK-Python/issues/156#issuecomment-624504822_

JonasVautherin, 2020-05-06:
Hello!

The error says that it cannot find `mavsdk_server`. One way to run an executable is to `cd` into the directory in which it is, and call it. Say you have the `mavsdk_server` binary in `/home/user/mavsdk/mavsdk_server`, it would be something like:

```
cd /home/user/mavsdk
./mavsdk_server serial:///dev/ttyACM0
```

Another way would be to install it on the system, such that it appears in the PATH, and directly run it as you mentioned above. But it really needs to be in the PATH in this case.

Safranek42, 2020-05-06:
Thanks for creating the new issue!

As I couldn't find the mavsdk_server executable,  I built and installed MAVSDK again with this (https://mavsdk.mavlink.io/develop/en/contributing/build.html) steps.

I checked and mavsdk_server is not among the Build targets. (Though there is mavsdk_server.cpp file after cloning the repository.)

I found this (https://github.com/mavlink/MAVSDK/issues/814) issue, but it was on Windows 10, and as I understand it is solved. (Also I do not get any error message.)

The MAVSDK folder is in home/username/Documents/. Where should it build the executables, where I could search for it?

Thanks for your help!

JonasVautherin, 2020-05-06:
> As I couldn't find the mavsdk_server executable

We release some artifacts here: https://github.com/mavlink/mavsdk/releases. But for the Jetson, you're probably right that we don't have a binary for that. So compiling from sources is the way to go. Possibly you want to compile from the latest release tag (instead of `develop`, which is less stable).

> I checked and mavsdk_server is not among the Build targets.

Not sure how you built, but you need to enable `BUILD_BACKEND` to get `mavsdk_server` (it is a bit unfortunate, but `mavsdk_server` used to be called "the backend" :sweat_smile:). Something like here: https://github.com/mavlink/MAVSDK/blob/develop/.github/workflows/main.yml#L63

```
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_BACKEND=ON -j 2 -Bbuild/release -H.
```

> The MAVSDK folder is in home/username/Documents/. Where should it build the executables, where I could search for it?

If you build in, say, `~/Documents/mavsdk/` with cmake and `-Bbuild/release`, it will go into `~/Documents/mavsdk/build/release/src/backend/src/mavsdk_server`.

Safranek42, 2020-05-07:
Many thanks, it worked, now I have the mavsdk_server executable and can connect to my flight controller!

JonasVautherin, 2020-05-07:
Awesome! :tada:

closed 2020-05-07 by JonasVautherin

MKexplorer, 2020-08-10:
I've installed mavsdk in my Ubuntu 18.04 through pip3 Install mavsdk.
But the mavsdk_server was not installed along with.
That's why I downloaded it manually.
Now I don't know what to do with the downloaded mavsdk_server file?
What I've to do to make it working when I'm working with MAVSDK Python?

JonasVautherin, 2020-08-10:
> But the mavsdk_server was not installed along with.

How do you know that? Note that pip will not put `mavsdk_server` in your PATH, so you won't be able to just run `$ mavsdk_server`, right?

MKexplorer, 2020-08-10:
Yes sir,
At first I cannot find the mavsdk_server in the location /usr/local/lib/dist-packages/Python 3.6/mavsdk/bin 
That's why I installed the server file at this exact location and make it executable.

But now I couldn't able to find any guide or docs which tells what to do next or how to use the mavsdk_server file?

Basically I'm trying to control a jmavsim sitl with MAVSDK python scripts but I couldn't able to connect the server when I simulate the Jmavsim.
I'm following this guide written by you https://auterion.com/getting-started-with-mavsdk-python/ 
And when I run await drone.connect() it shows waiting for mavsdk_server to be ready...
What to do now?
I hope I can able to make you understand about my issue.

Thanks for your reply!

JonasVautherin, 2020-08-11:
I don't really get why `mavsdk_server` doesn't come from `pip3` on Ubuntu 18.04. But let's start it manually:

Run `mavsdk_server` like so:

```
cd /path/to/mavsdk_server/directory
./mavsdk_server -p 50051 udp://:14550
```

Then connect MAVSDK-Python to it:

```python
from mavsdk import System

drone = System(mavsdk_server_address="127.0.0.1", port=50051)
await drone.connect()
```

MKexplorer, 2020-08-11:
Thanks for your reply.
But at first when I cd to the mavsdk_server file and run ./mavsdk_server ..... :14550, it shows this error!

./mavsdk_server: cannot execute binary file: Exec format error.
Why this error comes?

MKexplorer, 2020-08-11:
[image]

This one!

MKexplorer, 2020-08-11:
Thanks a lot sir for your valuable guidance.
Now it's perfectly working 🥳

MKexplorer, 2020-08-15:
Sir, again I'm facing a simple problem but not getting a workable solution.
I'm not able to upload a MAVSDK python mission script to the jmavsim simulator.
I got the answer for c++ scripts but not for Python scripts.

So how to/ what's the command to upload (eg.  https://github.com/mavlink/MAVSDK-Python/blob/0.3.0/examples/takeoff_and_land.py ) this script to the simulator?