mavlink/MAVSDK-Python #181: Telemtry logging fails

Safranek42, 2020-05-07:
Hi,

I ran into a problem using MAVSDK-Python telemetry class. I am able to initiate connection to my flight controller with mavsdk_server, and then also connect with examples/telemetry.py (every time I run it).

However, telemetry data is sometimes streamed, and sometimes not. So far I could not figure out anything systematic; it happens that I run the script, it works, I stop is, immediately run again and not working. I tried, reconnecting the flight control board, modifying `/dev/ttyACM1` permissions, and not defining the port for `mavsdk_server`, but rather use the port number of it in my telemetry python script; none of them solved the problem.

What seems to be happening is in 
`async for in_air in drone.telemetry.in_air():`
the `drone.telemetry.in_air()` object is "empty", so the program does not go into the for loop, but is constantly trying, since it runs asynchronously.

Anyone has any idea why it fails and runs correctly seemingly random?
Thanks for any help!

Specs:
My flight controller is Durandal.
Using PX4 master build
Running mavsdk-python on NVidia Jetson Nano board.
The Jetson and Durandal are connected via USB, serial /dev/ttyACM1:115200.
For mavsdk server I'm using the port option `-p 50051`.

JonasVautherin, 2020-05-07:
Just to make sure I understand:

* `examples/telemetry.py` always works
* your script does not always receive the telemetry

Correct?

One limitation that we currently have in Python (as opposed to Java or Swift) is that if you call `drone.telemetry.position()` twice, the second call will stop the first one. So you can have only one coroutine running an async generator (here say `drone.telemetry.position()`) at a time. I opened an issue [here](https://github.com/mavlink/MAVSDK-Python/issues/60) a while ago.

Could it be the issue you are experiencing?

Safranek42, 2020-05-08:
Hi,

thanks for your answer.

> * `examples/telemetry.py` always works
> * your script does not always receive the telemetry

Not exactly. The behavior of telemetry.py is the same as my script: sometimes works, sometimes not. The difference between the two script is mainly that I am writing to file instead of terminal. (I just wanted to test with the "official" example also.)

> One limitation that we currently have in Python (as opposed to Java or Swift) is that if you call `drone.telemetry.position()` twice, the second call will stop the first one. So you can have only [...]

I don't run two programs in the same time, so I don't think that could be a problem.

Today morning I tried again, and it is always running correctly. Which is strange as I haven't changed anything since yesterday, in the code or in the way I run it or use mavsdk_server. I will test it more and let you know the results.

JonasVautherin, 2020-05-08:
When you don't receive the telemetry, can you do something else (like `arm` and `takeoff`), or does everything stall? Because maybe the drone is not detected at all.

We should have a way to get more logs in python, I should have a look at that. So we would see if you are connected or not. One thing you can try is to start `mavsdk_server -p 50051` manually, and connect to that instance (with `drone = System(mavsdk_server_address=localhost)`). So that you'll get the log output of `mavsdk_server`.

Safranek42, 2020-05-08:
I don't do anything else, just getting telemetry.
I'm connecting the way you described. Since it is working now I cannot check the output of `mavsdk_server` in the 'wrong' case but if it happens I will save it.

The output when it is working is this:

```
$ ./mavsdk_server -p 50051 serial:///dev/ttyACM0:115200
[12:15:23|Info ] MAVSDK version: 0.24.0-1-ga517ee11 (mavsdk_impl.cpp:25)
[12:15:23|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:401)
[12:15:23|Info ] Server started (grpc_server.cpp:38)
[12:15:23|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:39)
[12:15:23|Info ] Waiting to discover system on serial:///dev/ttyACM0:115200... (connection_initiator.h:22)
[12:15:23|Debug] Component Autopilot (1) added. (system_impl.cpp:386)
[12:15:23|Debug] Discovered 1 component(s) (UUID: 3474616186718925878) (system_impl.cpp:556)
[12:15:23|Info ] System discovered [UUID: 3474616186718925878] (connection_initiator.h:61)
[12:15:24|Debug] MAVLink: warning: Preflight: GPS Horizontal Pos Drift too high (system_impl.cpp:292)

```

JonasVautherin, 2020-05-09:
Right. Please share the output next time it fails, then :blush:

julianoes, 2020-07-29:
Closing as there was no more answers here.

closed 2020-07-29 by julianoes