mavlink/MAVSDK-Python #189: Run MAVSDK-Python on a Jetson Nano

JonasVautherin, 2020-05-14:
When I ran TakeoffAndLand.py I was also getting the message telling me to manually build mavsdk myself. Because I installed it using pip3. So I built it according to the directions in the C++ section of mavsdk.
Couple questions 
1. when i run ./mavsdk_server where do I do it? I've tried to run it from linux home and inside the mavsdk folder and it doesnt exist
2.  Also im using a jetson nano to connect to a pixhawk through USB so the endpoint should be /dev/ttyUSB0:57600? So dronekit python will tell mavsdk_server to go look for a pixhawk connected through usb?
Thanks

_Originally posted by @maxwelllwang in https://github.com/mavlink/MAVSDK-Python/issues/155#issuecomment-628455936_

JonasVautherin, 2020-05-14:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mavsdk/system.py", line 166, in _start_mavsdk_server
    with path(bin, 'mavsdk_server') as backend:
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/usr/lib/python3.8/importlib/resources.py", line 201, in path
    with open_binary(package, resource) as fp:
  File "/usr/lib/python3.8/importlib/resources.py", line 91, in open_binary
    return reader.open_resource(resource)
  File "<frozen importlib._bootstrap_external>", line 988, in open_resource
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.8/dist-packages/mavsdk/bin/mavsdk_server'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "takeoff_and_land.py", line 37, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.8/asyncio/base_events.py", line 608, in run_until_complete
    return future.result()
  File "takeoff_and_land.py", line 10, in run
    await drone.connect(system_address="udp://:14540")
  File "/usr/local/lib/python3.8/dist-packages/mavsdk/system.py", line 62, in connect
    self._start_mavsdk_server(system_address)
  File "/usr/local/lib/python3.8/dist-packages/mavsdk/system.py", line 175, in _start_mavsdk_server
    raise Exception("\nIt seems like this installation does not provide an embedded 'mavsdk_server' binary. If you installed from pip, it means that 'mavsdk_server' is not distributed for your platform (yet). You will need to get and run it manually:\n\n\t1. Build 'mavsdk_server': https://github.com/mavlink/mavsdk.\n\t2. Run it, e.g. on port 50051: './mavsdk_server -p 50051'.\n\t3. Set the 'mavsdk_server_address' and port when creating the System: 'drone = System(mavsdk_server_address='localhost', port=50051)\n")
Exception: 
It seems like this installation does not provide an embedded 'mavsdk_server' binary. If you installed from pip, it means that 'mavsdk_server' is not distributed for your platform (yet). You will need to get and run it manually:

    1. Build 'mavsdk_server': https://github.com/mavlink/mavsdk.
    2. Run it, e.g. on port 50051: './mavsdk_server -p 50051'.
    3. Set the 'mavsdk_server_address' and port when creating the System: 'drone = System(mavsdk_server_address='localhost', port=50051)'
```

Is this because I installed mavsdk incorrectly? should I try removing it using pip and trying again?

_Originally posted by @maxwelllwang in https://github.com/mavlink/MAVSDK-Python/issues/155#issuecomment-628455936_

JonasVautherin, 2020-05-14:
> when i run ./mavsdk_server where do I do it?

Following the instructions, you built a `mavsdk_server` executable. That means that you ran something like `cmake --build build/default` somewhere, right? The built executable is therefore inside this `build/default` folder, normally in `build/default/src/backend/src/mavsdk_server`.

`./mavsdk_server`, in Linux, means "run the executable located in the current directory (`./`) and called "mavsdk_server". So you have to move to that directory in the command line (with e.g. `cd build/default/src/backend/src`).

> Also im using a jetson nano to connect to a pixhawk through USB so the endpoint should be /dev/ttyUSB0:57600

I guess you will get a serial connection, I cannot say exactly which name it will be. On my computer I typically get `/dev/ttyACM0`. When you plug the pixhawk, you may find that in the output of `dmesg`.

> So dronekit python will tell mavsdk_server to go look for a pixhawk connected through usb?

No dronekit involved here :wink:. MAVSDK-Python will start `mavsdk_server` and tell it to listen to a MAVLink connection on the address you give in `await drone.connect(system_address="<address>")`, which in your case would look like `serial:///dev/ttyUSB0:57600`.

maxwelllwang, 2020-05-15:
Thank you! Turns out I built it in the wrong place. And Oops on the dronekit-python typo I just recently switched to MAVSDK python 😓 
Now when I run examples It says waiting for mavsdk_server to be ready so I have to manually start the server with ./mavsdk_server -p 50051 Which I installed system wide. 
Is there anyway to start mavsdk_server automatically when I run an example? or worse case start automaically on bootup?
Thanks! :)

maxwelllwang, 2020-05-15:
Also I tried both connection strings "serial://dev/ttyUSB0:57600" and "serial://dev/ttyACM0:57600" Both leave me at " waiting for drone connect". I am connecting from jetson nano usb port to micro usb on pixhawk. I successfully from mavproxy but didn't specify a port and it found it automatically. However I can't replicate it anymore because for some reason python can't find mavproxy anymore.

JonasVautherin, 2020-05-15:
> And keeps giving you `command unsupported (528) mavlink_commans.cpp165` not sure what that means

As documented in the MAVLink specifications, command 528 is [MAV_CMD_REQUEST_FLIGHT_INFORMATION](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_FLIGHT_INFORMATION). I guess you're not using PX4, are you?

> Still would like to know how to start mavsdk_server on boot up or when python imports mavsdk

I would say that this is not specifically a MAVSDK issue (but rather linux/python generic questions), and would recommend you make a search for that (e.g. [here](https://duckduckgo.com/?q=linux+run+executable+on+boot&t=brave&ia=web) and [here](https://duckduckgo.com/?q=python+run+process&t=brave&ia=web) :wink:.

maxwelllwang, 2020-05-15:
Nope not using PX4

Ok I will look into the two links thanks.
But since I have to manually specify the connection address when i start mavsdk_server
the second line doesn't do anything? I feel that mavsdk_server and python arn't linking correctly
```
drone = System(mavsdk_server_address='localhost', port=50051)
await drone.connect(system_address="serial:///dev/ttyACM0:57600")
```

julianoes, 2020-05-15:
The `drone.connect` call can be used if you are using the pip package on Linux x86_64, Win32/64, macOS x64_64 but not on arm architectures because we can't ship the binary using pip ([yet](https://github.com/mavlink/MAVSDK/issues/997)).
On arm architetures you need to start `mavsdk_server` separately, as you did. Does that answer your question?

JonasVautherin, 2020-05-15:
> But since I have to manually specify the connection address when i start mavsdk_server
the second line doesn't do anything?

Good question. You always need both lines, but if `mavsdk_server_address` is specified in the first one, then `system_address` is ignored in the second one. In your case it would mean:

```python
drone = System(mavsdk_server_address='localhost', port=50051)
await drone.connect()
```

And `mavsdk_server` is started manually on 'localhost' with:

```
./mavsdk_server -p 50051 serial:///dev/ttyACM0:57600
```

JonasVautherin, 2020-07-15:
Feel free to reopen if you have more questions :blush:.

closed 2020-07-15 by JonasVautherin