mavlink/MAVSDK-Python #190: mavsdk 0.7.0 with PX4 1.11.0 - Mission Errors

breathedigital, 2020-05-17:
Hello -
Upgraded PX4 FW & MavSDK-Python to 1.11.0 80 & 0.7.0.

And, now we're going through classic "it worked on my computer, but not when deployed" syndrome.

Error:
````
UNKNOWN: ''; origin: set_return_to_launch_after_mission(); params: (True,)
````

How to reproduce:
Run the `mission.py` from `examples` directory on python 3.7

The error message is quite vague -- its going through UNKNOWN MISSION ERROR path. Any help is highly appreciated!

Cheers!
ram

JonasVautherin, 2020-05-18:
> Upgraded PX4 FW & MavSDK-Python to 1.11.0 80 & 0.7.0.

Are you running `mavsdk_server` manually, or are you using the embedded one (if you don't know, then it means that you use the embedded one, which is started automatically :laughing:).

If you run `mavsdk_server` manually, which version are you using? MAVSDK-Python 0.7.0 [requires `mavsdk_server` v0.26.0](https://github.com/mavlink/MAVSDK-Python/blob/0.7.0/MAVSDK_SERVER_VERSION).

breathedigital, 2020-05-19:
Appreciate your quick reaction! 

mavsdk_server was run manually @ -p 50051. 
Indeed, it was older than 0.26.0 causing this error.

Many Thanks! 🎇

_ps: A compatibility matrix of **mavsdk_server**, **mavsdk-python**, **px4 firmware** would ease things drastically._

closed 2020-05-19 by breathedigital