mavlink/MAVSDK-Python #191: Run MAVSDK-Python on Raspberry pi 3b

Raghava248, 2020-05-22:
I followed all the instructions specified here 
https://discuss.px4.io/t/talking-to-a-px4-fmu-with-a-rpi-via-serial-nousb/14119

but iam getting this error please help me 

pi@raspberrypi:~/MAVSDK-Python/examples $ python3 firmware_version.py
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 166, in _start_mavsdk_server
    with path(bin, 'mavsdk_server') as backend:
  File "/usr/lib/python3.7/contextlib.py", line 112, in __enter__
    return next(self.gen)
  File "/usr/lib/python3.7/importlib/resources.py", line 201, in path
    with open_binary(package, resource) as fp:
  File "/usr/lib/python3.7/importlib/resources.py", line 91, in open_binary
    return reader.open_resource(resource)
  File "<frozen importlib._bootstrap_external>", line 929, in open_resource
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/.local/lib/python3.7/site-packages/mavsdk/bin/mavsdk_server'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "firmware_version.py", line 23, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "firmware_version.py", line 9, in run
    await drone.connect(system_address="serial:///dev/serial0:921600")
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 62, in connect
    self._start_mavsdk_server(system_address)
  File "/home/pi/.local/lib/python3.7/site-packages/mavsdk/system.py", line 175, in _start_mavsdk_server
    raise Exception("\nIt seems like this installation does not provide an embedded 'mavsdk_server' binary. If you installed from pip, it means that 'mavsdk_server' is not distributed for your platform (yet). You will need to get and run it manually:\n\n\t1. Build 'mavsdk_server': https://github.com/mavlink/mavsdk.\n\t2. Run it, e.g. on port 50051: './mavsdk_server -p 50051'.\n\t3. Set the 'mavsdk_server_address' and port when creating the System: 'drone = System(mavsdk_server_address='localhost', port=50051)\n")
Exception:
It seems like this installation does not provide an embedded 'mavsdk_server' binary. If you installed from pip, it means that 'mavsdk_server' is not distributed for your platform (yet). You will need to get and run it manually:

        1. Build 'mavsdk_server': https://github.com/mavlink/mavsdk.
        2. Run it, e.g. on port 50051: './mavsdk_server -p 50051'.
        3. Set the 'mavsdk_server_address' and port when creating the System: 'drone = System(mavsdk_server_address='localhost', port=50051)

closed 2020-05-25 by Raghava248

KritinMathur, 2021-01-09:
Did u find the solution ?

JonasVautherin, 2021-01-09:
We now provide an embedded `mavsdk_server`, please try that and open a new issue if that's not working.

ytaskiran, 2021-01-16:
In the [documentation](https://mavsdk.mavlink.io/develop/en/getting_started/installation.html) there is a statement that points out the RPi is not supported for MAVSDK-Python.

> MAVSDK-Python is currently only available for Windows, macOS, and Linux x86_64. It is not yet suppported for Linux ARM platfroms like RPi 

This may lead to confusion about the availability. Actually, when I first read here, I thought I cannot use MAVSDK-Python with RPi at all. Maybe you can leave a not there to indicate there is a way to utilize MAVSDK-Python and install mavsdk-server.

JonasVautherin, 2021-01-16:
@ytaskiran: nice catch for the documentation! :blush: 

As can be seen [here](https://pypi.org/project/mavsdk/#files), we distribute mavsdk-python for armv6, armv7 and aarch64, which should cover the RPis.

Would you be willing to contribute this to the docs? You can just remove the note, it's outdated now :+1: :slightly_smiling_face:

ytaskiran, 2021-01-17:
Absolutely, I'd like to do. Will get it done as soon as possible.