mavlink/MAVSDK-Python #193: mavsdk_daemon - Heartbeats timing out!

ramgrandhi, 2020-05-23:
Hello -
On AARCH64 machine, I'm running mavsdk_server explicitly with the following parameters. To the USB port is where I connected PixHawk4 via USB<->Serial converter.

````
mavsdk_server -p 50051 serial:///dev/ttyUSB0:115200
````

mavsdk_server does detect the UUID of PX4 but it keeps losing heartbeats (timing out) and trying to reconnect ... and this loop goes on and on. It is not sounding reliable at all.

````
................................
[03:24:37|Info ] System timed out [UUID: 354667XXXXXXXXX847] (connection_initiator.h:40)
[03:24:38|Debug] Discovered 1 component(s) (UUID: 3546673858082977847) (system_impl.cpp:525)
[03:24:38|Warn ] sending again after 0.510021 s, retries to do: 3  (511). (mavlink_commands.cpp:224)
[03:24:39|Warn ] sending again, retries to do: 3  (CAL_GYRO0_ID). (mavlink_parameters.cpp:458)
[03:24:39|Warn ] sending again after 1.01386 s, retries to do: 2  (511). (mavlink_commands.cpp:224)
[03:24:39|Warn ] sending again after 1.52498 s, retries to do: 1  (511). (mavlink_commands.cpp:224)
[03:24:40|Error] Retrying failed (511) (mavlink_commands.cpp:243)
[03:24:46|Info ] heartbeats timed out (system_impl.cpp:262)
[03:24:46|Debug] Lost 354667XXXXXXXXX847 (mavsdk_impl.cpp:406)
[03:24:46|Info ] System timed out [UUID: 354667XXXXXXXXX847] (connection_initiator.h:40)
[03:24:46|Debug] Discovered 1 component(s) (UUID: 354667XXXXXXXXX847) (system_impl.cpp:525)
..........................
````

Can you suggest a few things that can be done to improve this reliability? Strangely, everything is wired together yet resulting in such high levels of timeouts, it is puzzling.

Appreciate it!

cheers,
ram

julianoes, 2020-05-25:
Hi ram,

Does it work in other configurations. E.g. does it work with QGC properly, or does it work running on a desktop?

And can you share the output of `mavlink status` from the PX4 side?

ramgrandhi, 2020-06-01:
Hi @julianoes -
Sorry for the delay.

After looking at your reply, I have a more generic question for you with respect to the compatibility-matrix between PX4/Firmware (and) MAVSDK (and) MAVSDK-Python libraries.

Here is my combination.
````
PX4/Firmware (stable 1.10.1) + MAVSDK (0.26.0) + MAVSDK-Python (0.7.0).
````
Question: Is this a reliable combination? Or, do you suggest changes.

Problem: With this setup, in SITL mode, I am having troubles with Barometer & GPS.

````
pxh> INFO  [ecl/EKF] 1532000: EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [ecl/EKF] 6092000: EKF commencing GPS fusion
WARN  [ecl/EKF] 6708000: EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] 11716000: EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] 26364000: EKF baro hgt timeout - reset to GPS
WARN  [ecl/EKF] 26364000: EKF gps hgt timeout - reset to GPS
````

Here is the output of `mavlink status` command.

````
pxh> mavlink status

instance #0:
	mavlink chan: #0
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 9.154 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Normal
	MAVLink version: 1
	transport protocol: UDP (14570, remote port: 14550)

instance #1:
	GCS heartbeat:	344000 us ago
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 23.370 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.270 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	MAVLink version: 2
	transport protocol: UDP (14580, remote port: 14540)
	partner IP: 127.0.0.1
````

Appreciate your time!

Cheers.

julianoes, 2020-06-02:
> Question: Is this a reliable combination? Or, do you suggest changes.

This should be ok, however, I would now recommend going to `MAVSDK (0.27.0)` + `MAVSDK-Python (0.8.0)` which fixes a few things we missed before.

> WARN  [ecl/EKF] 6708000: EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] 11716000: EKF baro hgt timeout - reset to baro
WARN  [ecl/EKF] 26364000: EKF baro hgt timeout - reset to GPS
WARN  [ecl/EKF] 26364000: EKF gps hgt timeout - reset to GPS

This is solely a PX4 issue, not sure why you are seeing it. I assume it has to do with what your simulating, the settings you are using, or changes you have done. If it persists, I'd create an issue for PX4/Firmware or ask in discuss.px4.io.

The `mavlink status` output is normal for SITL, however, your problem was with actual Pixhawk hardware over USB, so that's not helping for that case.

julianoes, 2021-03-26:
Closing this as there was no follow up. Please comment again if this is still an issue.

closed 2021-03-26 by julianoes