mavlink/MAVSDK-Python #197: ImportError: cannot import name 'start_mavlink'

clarknator, 2020-05-30:
I watched 45 min video here https://www.youtube.com/watch?v=VFT4P_N7wc8

I entered apython and received error when command was >>from mavsdk import start_mavlink

Do I need to generate a MAVSDK_server first?

JonasVautherin, 2020-05-31:
I think this is a bit outdated, sorry :see_no_evil:. I would advise you have a look at [this blogpost](https://auterion.com/getting-started-with-mavsdk-python/). This information should be in the official documentation, I'll try to bring it there ASAP.

I think that what you want to do from `apython` is to run more something like this (explained in the blog post), in this case to create the `drone` object and use it to arm the drone:

```python
from mavsdk import System
drone = System()
await drone.connect()
await drone.action.arm()
```

clarknator, 2020-05-31:
Thank you. I realized after I posted that this was closed. Warmest regards.

closed 2020-05-31 by clarknator

JonasVautherin, 2020-05-31:
Does it work now, with `from mavsdk import System`? :slightly_smiling_face:

sprucetonye, 2021-03-06:
No please I'm still getting an error message

JonasVautherin, 2021-03-07:
I am confused, @sprucetonye. That issue was closed last year, and it was opened by @clarknator.

Please open a new issue and describe your problem.