mavlink/MAVSDK-Python #198: Commands get Denied

tgadi001, 2020-05-31:
Hi,
I have been trying to write a code to make my drone takeoff or use the goto_location function. Unfortunately it always gives me a "Command Denied" response. I am not able to get_takeoff_altitude or set_takeoff_altitude either, but my arm function works just fine and I was getting necessary telemetries right up until latest release. I have been trying to run examples that involve takeoff, but none of them work.
So my question is why most of my Action functions are not able to run? Do you think i did something wrong when setting up the MAVSDK-Python?
I would really appreciate your help or any advice as i have been trying to fix this issue for couple month now. Thank you in advance!

JonasVautherin, 2020-05-31:
Are you using the embedded `mavsdk_server`, or are you running something like `./mavsdk_server -p 50051` manually?

Are you running against SITL?

Could you show the code you run that fails?

tgadi001, 2020-05-31:
I am pretty sure I am using the embedded version and I am running it against SITL.

`async def arm_drone(self, drone):

        print("arm_drone ran")
        await self.drone.connect(system_address="udp://:14551")

        await self.drone.action.arm()
        
        await self.drone.action.set_takeoff_altitude(20.0)

        await self.drone.action.takeoff()

        await self.drone.action.goto_location( -35.363261, 149.168372 , 120.0, 1.0)
        

        await asyncio.sleep(15)

        self.drone = None`

The code is pretty simple and looks similar to examples. The arm function successfully works with SITL.

JonasVautherin, 2020-05-31:
Did you try to run one of the examples, e.g. `takeoff_and_land.py` (changing the `system_address, since apparently you have `udp://:14551`)? Does that work?

Just to be sure: are you running on PX4? Or is it APM?

tgadi001, 2020-05-31:
running that example gives me this error
`-- Arming
-- Taking off
Action Takeoff started
ActionResult: [result: COMMAND_DENIED, result_str: Command denied]
Traceback (most recent call last):
  File "takeoff_and_land.py", line 50, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.8/asyncio/base_events.py", line 612, in run_until_complete
    return future.result()
  File "takeoff_and_land.py", line 40, in run
    await drone.action.takeoff()
  File "/home/name/.local/lib/python3.8/site-packages/mavsdk/generated/action.py", line 280, in takeoff
    raise ActionError(result, "takeoff()")
mavsdk.generated.action.ActionError: COMMAND_DENIED: 'Command denied'; origin: takeoff(); params: ()`
Sometimes when I try to run takeoff_and_land.py it breaks my SITL and causes the connection to be refused. In this case i have to restart SITL completely.

I am going to use Navio2, so i think it is APM. Right now i am writing my own version of ground control station software, but i just cant figure out what leads to the commands being denied.

JonasVautherin, 2020-05-31:
> I am going to use Navio2, so i think it is APM.

Well that may be related. We develop and test MAVSDK against PX4. An easy way to try MAVSDK against a PX4 SITL is described [here](https://github.com/jonasvautherin/px4-gazebo-headless). I would suggest you try that to make sure that MAVSDK works on your system. Would you mind trying that?

tgadi001, 2020-06-02:
Thank you so much, this actually helped a lot. I Finally i can continue developing! I guess that APM SITL is not very compatible with MAVSDK. I am looking forward for all the updates to come on the python wrapper. Thank you again!!

closed 2020-06-02 by tgadi001