mavlink/MAVSDK-Python #199: Cannot ARM using mavsdk commands but LAND works.

clarknator, 2020-06-01:
I have a custom GCS that works well with APM SITL. 
//----

**is there a way to use mavsdk with AruPilot?** 

//---

I am new to mavsdk and want to write python for auto regression tests. Simple await drone.action.arm() eludes me. The result is below...

Traceback (most recent call last):
  File "/home/rrr6399/.local/lib/python3.6/site-packages/aioconsole/execute.py", line 95, in aexec
    result, new_local = yield from coro
  File "<aexec>", line 2, in __corofn
  File "/home/rrr6399/.local/lib/python3.6/site-packages/mavsdk/generated/action.py", line 250, in arm
    raise ActionError(result, "arm()")
mavsdk.generated.action.ActionError: COMMAND_DENIED_LANDED_STATE_UNKNOWN: 'Command denied, landed state is unknown'; origin: arm(); params: ()
>>> 
However, I can arm and launch using my GCS, something I cannot do with mavsdk. I see results in the terminal where I started mavsdk_server as shown in file-A. 

[File-A.txt](https://github.com/mavlink/MAVSDK-Python/files/4708800/File-A.txt)

Here, I start the linux mavsdk_server using port 14550. Then I use my GCS to arm and launch. This raises the hexacopter to 3m. 

For landing, I could use my GCS but instead, I tried the mavsdk command "await drone.action.land()". It worked. My GCS verified landing. 

Is mavsdk exclusive to PX4 and will not work with ArduPilot? Could the differences be so great as to stop fundamental actions like ARM  when using APM? Why else would action.land() work?

If true, is there a way to use mavsdk with APM? 

**Would building mavsdk_server instead of using the pre-built linux 64 file resolve my issue?** 

Please reference my setup as:
1. GCS running on iMac with connection 0.0.0.0:14550
2. SITL APM running on VM Ubuntu 18 as localhost or 127.0.0.1
3. mavsdk_server running on same VM, called by $ ./mavsdk_server_manylinux1-x64 udp://127.0.0.1:14540
4. drone connected as....>>> drone = System(mavsdk_server_address='localhost', port=50051)

**In summary,** 

I start mavsdk_server with pre-built linux64 using udp://127.0.01:14540. Then using $apython, 
>>> from mavsdk import System
>>> drone= System(mavsdk_server_address='localhost', port 50051)
>>> await drone.connect()
**then the following gives traceback error that implies "COMMAND DENIED"**
>>> await drone.action.arm()
**but if I raise with my GCS, then the following works**
>>> await drone.action.land()

I like the concept of MAVSDK and would like to write regression test routines with it. 

Cheers!

JonasVautherin, 2020-06-01:
> is there a way to use mavsdk with ArduPilot?

We currently don't have any active contributors from the ArduPilot community, and it's not completely clear to me what would need to be done there. Currently MAVSDK is developed and tested against PX4...

JonasVautherin, 2020-07-15:
Closing this, feel free to reopen if needed. ArduPilot support can be discussed here: https://github.com/mavlink/MAVSDK/issues/728

closed 2020-07-15 by JonasVautherin