mavlink/MAVSDK-Python #200: mission.py example broken (type object 'MissionItem' has no attribute 'CameraAction')

petergerten, 2020-06-01:
Trying to execute the mission.py example I get:
```
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 255
Traceback (most recent call last):
  File "mission.py", line 103, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "mission.py", line 32, in run
    MissionItem.CameraAction.NONE,
AttributeError: type object 'MissionItem' has no attribute 'CameraAction'
```
I was able to reproduce this in a clean environment - Dockerfile:
```
FROM ubuntu:20.04
RUN apt-get update && apt-get install -y python3-pip git vim
RUN pip3 install mavsdk
```

julianoes, 2020-06-02:
@JonasVautherin could that be the problem you discovered and fixed in https://github.com/mavlink/MAVSDK-Python/pull/194 ? If so it should be fixed in 0.8.0.

JonasVautherin, 2020-06-02:
Yes it could be. @petergerten would you mind trying with 0.8.x?

petergerten, 2020-06-03:
@JonasVautherin I was actually using 0.8.0.
Just upgraded to 0.8.2 and the same issue remains the same.

PR #207 (2020-06-04): Make modules out of the plugins
An issue we have in the current system (where all the files are generated in `mavsdk/generated`) is that we don't have any namespacing between them. Therefore, `from mavsdk import MissionItem` could import `MissionItem` from `Mission`, or from `MissionRaw`. In this case, it broke the `examples/mission.py` example.

I therefore made a module out of each plugin, so that `import mavsdk.mission` is a thing, and one could `from mavsdk.mission_raw import MissionItem as MissionRawItem`, if needed.

I had to make a few changes for that:
* Remove the `globals()` use in ` [... cut]
closed 2020-06-04 by julianoes