mavlink/MAVSDK-Python #201: import_qgroundcontrol_mission() does not work (AttributeError: 'RepeatedCompositeFieldContainer' object has no attribute 'latitude_deg')

petergerten, 2020-06-01:
When using the following code to import a QGC plan:
`mission_items = await drone.mission.import_qgroundcontrol_mission("./test.plan")`

I get the following error:
```
  File "/usr/lib/python3.7/site-packages/mavsdk/generated/mission.py", line 1001, in import_qgroundcontrol_mission                                                                                   
    return MissionPlan.translate_from_rpc(response.mission_plan)                                  
  File "/usr/lib/python3.7/site-packages/mavsdk/generated/mission.py", line 325, in translate_from_rpc                                                                                               
    MissionItem.translate_from_rpc(rpcMissionPlan.mission_items)                                  
  File "/usr/lib/python3.7/site-packages/mavsdk/generated/mission.py", line 186, in translate_from_rpc                                                                                               
    rpcMissionItem.latitude_deg,                                                                  
AttributeError: 'RepeatedCompositeFieldContainer' object has no attribute 'latitude_deg' 
```

test.plan is a very simple mission exported via QGC 4.0.7:
https://pastebin.com/p8FRAWmU

julianoes, 2020-06-02:
Interesting, thanks for the issue.

JonasVautherin, 2020-06-03:
So it's failing [here](https://github.com/mavlink/MAVSDK-Python/blob/master/mavsdk/generated/mission.py#L186), because the generated code passes a repeated type [here](https://github.com/mavlink/MAVSDK-Python/blob/master/mavsdk/generated/mission.py#L325):

```python
return MissionPlan(
    MissionItem.translate_from_rpc(rpcMissionPlan.mission_items)
    )
```

And instead it should create a list of translated items around line 324. I'll have a look at the templates ASAP.

PR #206 (2020-06-03): Handle repeated non-primitive types in translate_from_rpc
Fixes #201.

@petergerten: would you mind testing this? On my end I can now load your `test.plan`.
closed 2020-06-03 by julianoes