mavlink/MAVSDK-Python #211: API Reference documentation seems to be broken

jank324, 2020-06-05:
The [API Reference documentation](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com) only shows content on the first page. Any of the pages only show the title, but no further content.

[Example](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/mission.html)
[image]

I did also notice that the documentation says it's for version 0.9.0 even though pip installs 0.8.2. I'm not sure that might have something to do with it.

julianoes, 2020-06-06:
Thanks for the not. I thought I had fixed it but this doesn't look like it.
0.9.0 is in fact the new one released on PyPi but it might not be depending on your platform, e.g. on arm.

julianoes, 2020-06-06:
It works locally but fails in CI:
https://travis-ci.org/github/mavlink/MAVSDK-Python/jobs/694956963#L629-L631

PR #212 (2020-06-06): sphinx: fix path to create docs
Fixes #211.
closed 2020-06-06 by julianoes