mavlink/MAVSDK-Python #216: ImportError after upgrading to most recent version

KonRoz, 2020-06-09:
Hi,

I ran into some some trouble after upgrading mavsdk from 0.8.0 to 0.9.1 for use in a GCS on Ubuntu 16.04 with python 3.8.
 
0.9.1

`(.venv) konrad@konrad-XPS-13-9370:~/Dronology-GCS$ python dronology/main.py 
Traceback (most recent call last):
  File "dronology/main.py", line 8, in <module>
    from dronology.Px4Drone import get_instance, loop
  File "/home/konrad/Dronology-GCS/dronology/Px4Drone.py", line 9, in <module>
    from mavsdk import MissionItem, MissionPlan
ImportError: cannot import name 'MissionItem' from 'mavsdk' (/home/konrad/Dronology-GCS/.venv/lib/python3.8/site-packages/mavsdk/__init__.py)`

0.8.0

`(.venv) konrad@konrad-XPS-13-9370:~/Dronology-GCS$ python dronology/main.py 
2020-06-09 13:56:34,729 | runtime  | INFO     | main.py              | line:53   | Connecting to 1 drone(s)
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!`

JonasVautherin, 2020-06-09:
Yes the imports have changed a bit. Can you show your file? At least `Px4Drone.py` line 9

julianoes, 2020-06-09:
Check this:

https://github.com/mavlink/MAVSDK-Python/blob/72c26ffe1991176f03af5942b8c17309a8582cfa/examples/mission.py#L5-L6