mavlink/MAVSDK-Python #217: timeout on command arm, takeoff etc.

Owiec94, 2020-06-09:
I'm getting an error while using example 'takeoff_and_land.py":

Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 1
-- Arming
Traceback (most recent call last):
  File "test.py", line 37, in <module>
    loop.run_until_complete(run())
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python37\lib\asyncio\base_events.py", line 587, in run_until_complete
    return future.result()
  File "test.py", line 25, in run
    await drone.action.arm()
  File "C:\Users\Admin\AppData\Local\Programs\Python\Python37\lib\site-packages\mavsdk\generated\action.py", line 230, in arm
    raise ActionError(result, "arm()")
mavsdk.generated.action.ActionError: TIMEOUT: 'Timeout'; origin: arm(); params: ()

What cause that problem?

JonasVautherin, 2020-06-10:
Are you using PX4?

Owiec94, 2020-06-10:
Thank You for Your quick answer!
Yes, I'm using Pixhawk 2.4.8 with arducopter v 4.0.3 connected through esp8266 (firmware firmware_UDP_2016-12-05.bin from ugcs). I can fully manage by this connection on Mission Planner, but using python example is problematic for it

JonasVautherin, 2020-06-10:
MAVSDK is developed and tested against PX4, so I cannot really say what is the expected behavior with Arducopter. I'm a bit surprised that `arm()` fails already. I guess it may be related to the flight mode: https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/action/action_impl.cpp#L168-L200

But I would need input from somebody who knows how APM works...

julianoes, 2020-06-10:
Ok, you're using ArduCopter, not PX4. MAVSDK is currently not compatible with ArduPilot:
See https://github.com/mavlink/MAVSDK/issues/728.

Owiec94, 2020-06-10:
If I understood good, mavsdk will not work with my FC https://static2.abc-rc.pl/pol_pl_Kontroler-lotu-Pixhawk-PX4-V2-4-8-karta-SD-1GB-Autopilot-PX4-8106_3.jpg
? If yes, is there any possibility to change firmware to working with this FC? Which FC are You using to work with mavsdk?
Or do You know any other possibility to create planning mission remotely and run without transmitter, just using ?

julianoes, 2020-06-10:
@Owiec94 you can use [QGroundControl](http://qgroundcontrol.com/) and then flash PX4 which is compatible with MAVSDK. The hardware itself works with both, ArduPilot or PX4.

Owiec94, 2020-06-10:
@julianoes  Thank You for Your answer, Actually I've reflashed my FC with QGroundControl software but it wont connect with that software. Can You help me?
What is weird - the other FC (the same model 2.4.8) with arducopter firmware can communicate with QGroundControl, but PX4 firmware no.

julianoes, 2020-06-10:
> wont connect with that software

What do you see?

Owiec94, 2020-06-10:
https://easyupload.io/71ug8o

https://easyupload.io/7cizef

https://easyupload.io/csi25f

Yellow message Your vehicle is not responding...
the blue led is blinking and the main strong led is turned off.

julianoes, 2020-06-10:
Ok, that's odd. Does it also not appear after you unplug USB and plug it back in?

Owiec94, 2020-06-10:
Sadly, no, tried all possible things, I've uploaded also firmware into esp8266 to connect via UDP but it still without any success.
Should I install any additional drivers etc? Notice: I've no problem with connecting with Mission Planner and QGroundControl with arducopter firmware.
But I want to use mavsdk with PX4 firmware.

julianoes, 2020-06-11:
@Owiec94 where did you buy the Pixhawk from? My suspicion is that it could be a (bad) clone that lacks some hardware component which prevents PX4 from starting properly. If you have a debug cable as explained in https://docs.px4.io/master/en/flight_controller/pixhawk.html#console-port you could try to see the startup messages and see if that shows any specific errors.

Owiec94, 2020-06-12:
Dear @julianoes Thank You for Your support. It was an error "avionics power low" 4.89V what is an mistake, my pixhawk power module has 5.3V output. Where I can calibrate or set lower threshold or do something to ignore that message?

julianoes, 2020-06-15:
Aha! There is a param to skip that check, I think it's this one: https://dev.px4.io/master/en/advanced/parameter_reference.html#CBRK_SUPPLY_CHK

closed 2020-06-15 by julianoes

end-of-small-sanctuary, 2025-03-28:
I'm having this same issue as well, i set up preflight checks within the code to show that the drone is armable (it arms successfully within qgroundcontrol, it hovers when i use the rc), but it refuses to arm using python.

here's my code:
```python

import asyncio
from mavsdk import System

async def run():
    """Main function to control drone operations."""
    try:
        # Initialize and connect to drone
        drone = await initialize_drone()
        
        # Perform pre-flight checks
        if not await pre_flight_checks(drone):
            print("Pre-flight checks failed. Aborting.")
            return

        # Execute flight sequence
        await execute_flight(drone)
        
    except Exception as e:
        print(f"Critical error occurred: {e}")
    finally:
        print("Flight sequence completed.")

async def initialize_drone() -> System:
    """Initialize and connect to the drone."""
    drone = System(mavsdk_server_address='127.0.0.1', port=50051)
    print("MAVSDK System initialized.")

    try:
        await drone.connect(system_address="udp://:14540")
        print("Waiting for drone connection...")
        
        # Wait for connection
        async for state in drone.core.connection_state():
            if state.is_connected:
                print("Connected to drone!")
                return drone
                
    except Exception as e:
        print(f"Connection failed: {e}")
        raise

async def pre_flight_checks(drone: System) -> bool:
    """Perform all necessary pre-flight checks."""
    try:
        print("\nRunning pre-flight checks:")
        
        # Check global position
        print("Waiting for global position...")
        async for health in drone.telemetry.health():
            if health.is_global_position_ok and health.is_home_position_ok:
                print("Global position OK")
                break
        
        # Check if system is armable
        async for health in drone.telemetry.health():
            print(f"System health: {health}")
            if health.is_armable:
                break
            else:
                print("System not armable!")
                return False
        
        # Check RC status
        print("Checking RC status...")
        async for rc_status in drone.telemetry.rc_status():
            print(f"RC Status: {rc_status}")
            if rc_status.is_available:
                break
            else:
                print("RC not available!")
                return False
                
        return True
        
    except Exception as e:
        print(f"Pre-flight check error: {e}")
        return False

async def execute_flight(drone: System):
    """Execute the flight sequence."""
    try:
  
        # Arm the drone
        if not await arm_drone(drone):
            raise Exception("Failed to arm drone")
            
        # Takeoff and hover
        print("Taking off...")
        await drone.action.takeoff()
        
        # Hover for 10 seconds
        print("Hovering...")
        await asyncio.sleep(10)
        
        # Land
        print("Landing...")
        await drone.action.land()
        
    except Exception as e:
        print(f"Flight error: {e}")
        raise

async def arm_drone(drone: System, max_attempts: int = 3) -> bool:
    """Attempt to arm the drone with retries."""
    for attempt in range(max_attempts):
        try:
            print(f"\nArming attempt {attempt + 1}/{max_attempts}...")
            await drone.action.arm()
            print("Arming successful!")
            return True
        except Exception as e:
            print(f"Arming failed: {e}")
            if attempt < max_attempts - 1:
                await asyncio.sleep(2)
    return False

if __name__ == "__main__":
    asyncio.run(run())
``` 

with the output:
Running pre-flight checks:
Waiting for global position...
Global position OK
System health: Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: True, is_global_position_ok: True, is_home_position_ok: True, is_armable: True]
Checking RC status...
RC Status: RcStatus: [was_available_once: True, is_available: True, signal_strength_percent: 41.0]

Skipping parameter configuration (not in MAVSDK v3.0.1)

Arming attempt 1/3...
Arming failed: TIMEOUT: 'Timeout'; origin: arm(); params: ()
and so on until it fails and stops running.

any help would be greatly appreciated!

julianoes, 2025-04-01:
@end-of-small-sanctuary please don't necrobump and create a new issue.

And while you're at it, do you also have the output logs of the mavsdk_server part. You can run it separately as explained here: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues