mavlink/MAVSDK-Python #218: display the RC channel input value on the screen.

junhyuk9976, 2020-06-10:
I want to display the RC channel input value on the screen.
I couldn't find it in the API reference.
Any help would be greatly appreciated.

julianoes, 2020-06-10:
Yes, it's currently not implemented looking at http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html.

Which MAVLink message information would the information be in that you need?

junhyuk9976, 2020-06-10:
HIL_RC_INPUTS_RAW(#92).

julianoes, 2020-06-10:
But this message is not implemented or used for PX4. It is specfically for HIL (hardware in the loop) simulation. Are you sure?

junhyuk9976, 2020-06-11:
> But this message is not implemented or used for PX4. It is specfically for HIL (hardware in the loop) simulation. Are you sure?

I couldn't explain it properly.
I am wondering if there is a function similar to HIL_RC_INPUTS_RAW(#92) in MAVSDK-API.

julianoes, 2020-06-11:
Do you want to send RC channels using MAVSDK or do you want to have telemetry about what is being sent by some other RC?

junhyuk9976, 2020-06-12:
I want to know the Hz value of the RC channel remotely.

like PyMAVLink

`import time
from pymavlink import mavutil

Drone = mavutil.mavlink_connection('udpin:192.168.1.1:14550')

while True:
    try:
        print(Drone.recv_match(type='RC_CHANNELS', blocking=True))
    except:
        pass
    time.sleep(0.1)`

> Do you want to send RC channels using MAVSDK or do you want to have telemetry about what is being sent by some other RC?

junhyuk9976, 2020-06-12:
I used the Insert code function, but it doesn't work..

julianoes, 2020-06-12:
Ok, you need telemetry for RC_CHANNELS, now I understand it.

slucasm, 2020-11-26:
Hi @julianoes,

How is going this feature? I am developing a safe landing system with computer vision and I need to detect the RC switches values to be able to abort the landing if the pilot wants.

Do you know any available method to do this with mavsdk or do I have to wait for this improvement?

Thank you in advance,

Sergi.

Nbelles, 2022-04-01:
Hey @julianoes,

I would also like to know the status of this feature request. I am trying to get the status of the RC switches as well. Thanks!

eduardkieser, 2022-05-24:
Hi, same question/request.
I have a Cube connected to a linux companion computer via the Telemetry uart and I would like to read the RC channel values that are sent from the transmitter on the ground on the companion computer. What is the status of this feature, is it possible? If not, what is the least painfull work-around?

JonasVautherin, 2022-05-24:
I guess we need somebody to contribute `RC_CHANNELS` to the `telemetry` plugin. Happy to help if someone is willing to do that :+1: