mavlink/MAVSDK-Python #220: How to connect MAVSDK with PixHawk 4 instead of Gazebo/Jmavsim simulator?

MuhammadBilal1, 2020-06-15:
@julianoes @JonasVautherin 
I setup the environment as you mentioned in the below guide of your's. And it worked absolutely fine. I ran the example you provided [here](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/takeoff_and_land.py). Example also ran successfully.

[Getting started with MAVSDK – Python](https://auterion.com/getting-started-with-mavsdk-python/)

Now my problem is working with actual PixHawk. After finishing the example and simulator work, I closed everything, rebooted my Ubuntu 18.04 machine.
Open the terminal window, connected my PixHawk 4 (PX4 is installed over it previously via QGroundControl, PixHawk is connected with system via USB cable) and ran the example (takeoff_and_land.py) but its stuck at **Waiting for drone to connect...** not going beyond this. Please see attached image below.

[image]

I am running Ubuntu 18.04 over a VM Virtual Machine.
I don't know much about the UDP and the Serial Ports things. I read about them [here](https://github.com/mavlink/MAVSDK-Python/issues/159#issuecomment-594034558).

JonasVautherin, 2020-06-15:
You are doing `await drone.connect(system_address="udp://:14540")`, and therefore `mavsdk_server` waits for UDP packets coming on port `14540`. Though your pixhawk is connected over serial over USB.

`$ dmesg` should show you the name of the device when you plug the pixhawk in. For me I usually get it under `/dev/ttyACM0` or `/dev/ttyACM1`. And I connect to it with e.g.:

```python
await drone.connect(system_address="serial:///dev/ttyACM0")
```

I hope this helps!

MuhammadBilal1, 2020-06-15:
> You are doing `await drone.connect(system_address="udp://:14540")`, and therefore `mavsdk_server` waits for UDP packets coming on port `14540`. Though your pixhawk is connected over serial over [...]

@JonasVautherin  Thanks for responding.
I tried your suggestions, but still I am getting an error. Please have a look in the attached image.
Am I doing something wrong!

[image]

MuhammadBilal1, 2020-06-15:
@JonasVautherin : Your suggestion worked. Thanks.

closed 2020-06-15 by MuhammadBilal1

Mumtazeagle, 2024-07-27:
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone = System()
    # await drone.connect(system_address="serial:dev/ttyACM0")
    await drone.connect(system_address="serial:///dev/ttyACM0")
    print("Connected to MAVSDK")

    # drone = System(mavsdk_server_address='localhost', port=50051)

    status_text_task = asyncio.ensure_future(print_status_text(drone))

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position estimate OK")
            break

    print("-- Arming")
    await drone.action.arm()

    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(10)

    print("-- Landing")
    await drone.action.land()

    status_text_task.cancel()

async def print_status_text(drone):
    try:
        async for status_text in drone.telemetry.status_text():
            print(f"Status: {status_text.type}: {status_text.text}")
    except asyncio.CancelledError:
        return

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())
    
    
    
    I wanted to cnnect my pixhawk to this python SDK script but it is not working

julianoes, 2024-07-27:
@Mumtazeagle please don't necrobump. Open a new issue and give all required information. And check out http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/index.html#debug-connection-issues.