mavlink/MAVSDK-Python #221: mavsdk_server is not installed in Docker image

jank324, 2020-06-23:
When I install MAVSDK-Python through pip in a Docker image, the mavsdk_server binary is not downloaded, even though I believe it should be (`sys.platform.startswith('linux') and platform.machine() == "x86_64"` from `setup.py` evaluates to `True`). Manually downloading the binary from `https://github.com/mavlink/MAVSDK/releases/download/v0.27.0/mavsdk_server_manylinux2010-x64` to `/usr/local/lib/python3.6/dist-packages/mavsdk/bin/mavsdk_server`, then setting it executable using `chmod +x` works perfectly and as if the binary had been installed by pip.

I am working off of the [px4io/px4-dev-simulation-bionic](https://hub.docker.com/r/px4io/px4-dev-simulation-bionic) Docker image. A minimal Dockerfile to reproduce this issue looks as follows:

```
FROM px4io/px4-dev-simulation-bionic
RUN pip3 install mavsdk
```

I have no idea why exactly this happens. But I'm assuming it should be an easy fix as it can be done manually no problem at all. Maintainability of the manual version is kind of bad, though.

JonasVautherin, 2020-06-23:
I just tried in an `ubuntu:20.04` with docker, and it seems to work. Would you mind trying that, too? I did the following:

```
docker run --rm -it ubuntu:20.04
apt update
apt install python3
apt install python3-pip
pip3 install mavsdk
```

I got the following output, confirming that it installed the `manylinux2010` wheel:

```
root@901a74d4b60f:/# pip3 install mavsdk
Collecting mavsdk
  Downloading mavsdk-0.9.1-py3-none-manylinux2010_x86_64.whl (7.1 MB)
     |████████████████████████████████| 7.1 MB 5.1 MB/s
```

And I could run the embedded `mavsdk_server` manually:

```
root@901a74d4b60f:/# /usr/local/lib/python3.8/dist-packages/mavsdk/bin/mavsdk_server  
[12:23:02|Info ] MAVSDK version: 0.27.0 (mavsdk_impl.cpp:26)
[12:23:02|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:372)
[12:23:02|Info ] Server started (grpc_server.cpp:43)
[12:23:02|Info ] Server set to listen on 0.0.0.0:44155 (grpc_server.cpp:44)
[12:23:02|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:22)
```

Could you confirm that it works for an `ubuntu:20.04` container?

jank324, 2020-06-23:
I can confirm that it does indeed work for an `ubuntu:20.04` container.

I also tried the exact same based on an `ubuntu:bionic` container (which the `px4io/px4-dev-simulation-bionic`container is based on). In this case it does not work. 

A few things to note:

- With `ubuntu:bionic` I get Python 3.6.9 (as opposed to 3.8.2)
- `pip3 install mavsdk` runs for the better part of 15 minutes in the `ubuntu:bionic` container and remains stuck on `Running setup.py bdist_wheel for grpcio ...`. I have actually seen this happen before as well.
- The following output seems to confirm that the wrong wheel is being installed:
```
Collecting mavsdk
  Downloading https://files.pythonhosted.org/packages/25/7f/d4f8a506fc6dc0660543a49024ef7bd7d63d3825d800c0722dcac056d918/mavsdk-0.9.1-py3-none-any.whl (192kB)
```

JonasVautherin, 2020-06-23:
Right. I don't know why, then. Note that `setup.py` is there to create the wheel, but it is not the one deciding which wheel is installed on a platform. This is done through the name: `mavsdk-0.9.1-py3-none-manylinux2010_x86_64.whl` says that it is a `manylinux2010_x86_64` wheel, and then I believe that `pip` is the one deciding which one it wants.

It's supposed to take the more specific one first (in this case `manylinux2010_x86_64`). So apparently, with bionic, `pip` seems to believe that the platform does not support `manylinux2010_x86_64` and it therefore falls back to `none`, which is the pure python package (i.e. the one that does not embed `mavsdk_server`).

I believe that it works (i.e. it downloads the `manylinux2010` one) on Ubuntu 18.04 (without docker), but I'm not completely sure. If somebody with an Ubuntu 18.04 could test that, it would be informative.

Some ideas regarding your notes:

> With ubuntu:bionic I get Python 3.6.9 (as opposed to 3.8.2)

This should be fine, we need 3.6+

> remains stuck on Running setup.py bdist_wheel for grpcio ...

I believe that if `pip` doesn't find binaries for the platform, it will try to build gRPC from sources. Which takes a while. This does not happen for mavsdk because we don't instruct `pip` on how to build `mavsdk_server`. Either it finds the binary, or it takes the pure wheel. But gRPC does. That's typically what happens when installing on an RPi: `pip install mavsdk` will end up building gRPC from sources.

> The following output seems to confirm that the wrong wheel is being installed

Yes, it chose the generic, "pure python" wheel.

julianoes, 2021-03-26:
Closing as it is should now be easy to install using `pip install mavsdk`.

closed 2021-03-26 by julianoes