mavlink/MAVSDK-Python #225: Connecting to drone

KatrijnChys, 2020-06-24:
I'm currently having some troubles with connecting the drone to my code. I'm not able to run any scripts form examples. For example when running the mission.py script, I don't get any error's. I only get "Waiting for mavsdk_server to be ready...".

I need to add the gRPC port mavsdk_server -p 50051.  
How do I do that?

JonasVautherin, 2020-06-24:
How did you install MAVSDK-Python? `pip3 install mavsdk`?

On which system are you running? Windows? macOS? Raspberry Pi?

KatrijnChys, 2020-06-25:
I use windows and installed mavsdk trough pycharm

JonasVautherin, 2020-06-25:
And what MAVLink system are you trying to connect it to? SITL running on UDP 14540?

KatrijnChys, 2020-06-25:
yes, I use this code to connect

>  drone = System()
    await drone.connect(system_address="udp://:14540")

>

julianoes, 2020-06-25:
Can you paste the full output? And are you starting `mavsdk_server -p 50051` on the side?

KatrijnChys, 2020-06-25:
the output I get is 
`Waiting for mavsdk_server to be ready...`

I don' think I'm starting something besides the code, but I'm not quite sure how servers work.

julianoes, 2020-06-26:
ok, which version do you have installed? I have never installed it using PyCharm, so I'm not sure what that would do. I usually do `pip3 install --user mavsdk`.