mavlink/MAVSDK-Python #226: Sending commands to vehicle with specific mav_sys_id

keremalatepe, 2020-06-27:
Hi, 
I am trying to control multiple vehicles which works in SITL (with docker on MacOS) that stated in [with](https://auterion.com/getting-started-with-mavsdk-python/) [.](https://github.com/jonasvautherin/px4-gazebo-headless) After I am starting 2 vehicles in SITL, I am changing MAV_SYS_ID parameter and output port of one of them. I know that MAVSDK-Python works with MAVLINK protocol but is there any chance to control vehicle with specific  MAV_SYS_ID? I am asking this because, both of my vehicles have same UUID. You can see my code example attached and it is slightly different from example file named "takeoff_and_land.py"
[image] . In this example of code, when I run it, both devices seen connected but arm and takeoff commands only given for first written device. I searched for answer in this Issues section but i couldn't find anything related. I examine source files but I couldn't find the place where I should change. Only there is an issue #102 but it is not done yet as I understood. 
Thank you for your answer.

julianoes, 2020-06-30:
Thanks for the question. I hope we can get back to you soon.

JonasVautherin, 2020-06-30:
> when I run it, both devices seen connected but arm and takeoff commands only given for first written device. 

Yes, that's exactly what the code should be doing. You do:

```python
await drone1.action.arm()
await drone1.action.takeoff()
await asyncio.sleep(5)
await drone1.action.land()
```

Which is asking drone1 to arm, takeoff and land 5 seconds later. If you want both drones to do that, you should do something like:

```python
await drone1.action.arm()
await drone2.action.arm()

await drone1.action.takeoff()
await drone2.action.takeoff()

await asyncio.sleep(5)

await drone1.action.land()
await drone2.action.land()
```

Does that make sense?

keremalatepe, 2020-06-30:
Hey, firstly, thank you for your responses. I am sorry, I just notice that I sent screenshot with missing part. Below this piece of code, I already did same commands for drone2. Problem in here as i said, UUID's of both device seen same and even though I am giving commands to drone2, system sends it to the drone1. Because it thinks that they are same device in my opinion. 

> I am trying to control multiple vehicles which works in SITL (with docker on MacOS) that stated in [with](https://auterion.com/getting-started-with-mavsdk-python/) [...]

As I stated here, I am using docker image that pushed to github by you -which really helped me to run SITL on my device- and when I am running twice this docker image in 2 terminals, they are having same UUID and this leads to misunderstanding from mavlink side as I understand. 

I don't know if I could express my problem but I hope I did. 

Thank you for your answer again, have a nice day.

JonasVautherin, 2020-07-09:
The UUID is an issue if both run on the same UDP connection. But that's not your case here: because you run two instances of `mavsdk_server`, only one of them can bind to e.g. `14540`. So you need to tell both instances to listen to a different MAVLink port (and also gRPC port), e.g.:

```
./mavsdk_server -p 50050 udp://:14540
./mavsdk_server -p 50051 udp://:14541
```

Then, in the python code, you need to connect the two `System`s to the right instance of `mavsdk_server`:

```python
drone1 = System(mavsdk_server_address="localhost", port=50050)
await drone1.connect()
drone2 = System(mavsdk_server_address="localhost", port=50051)
await drone2.connect()
```

Finally, you need your two SITL instances to broadcast on a different port. One on 14540 (drone1), and the other one on 14541 (drone2). My `px4-gazebo-headless` image doesn't do that, though. But there are scripts in PX4/Firmware to run multiple drones, and that should do just that :+1:.