mavlink/MAVSDK-Python #227: Intermittent connection exceptions

lachlanbrewster, 2020-06-28:
I am getting intermittent connection exceptions raised when trying to either get the current status of the drone and when trying to send commands (velocity body commands). The general structure of how I go about this is below. I want to be able to operate the drone while having tasks run in the background updating my drone object. 

`
async def background_tasks():

    asyncio.ensure_future(drone.in_air_log())
    ...
    # other tasks ...

async def in_air_log(self):
    # This code is within my drone class
    async for in_air in self.drone.telemetry.in_air():
        self.in_air = in_air

loop.run_until_complete(background_tasks())

# connect, arm, takeoff etc

while True:
    # do stuff
    loop.run_until_complete(drone.send_velocity(command_x, command_y, command_z))

`

I don't use  'await drone_command()'  in the main loop as I have a decent amount of code elsewhere which executes here displaying a web app which is not asynchronous.

I get the same problem regardless of if I use my background tasks method, or by putting 
'loop.run_until_complete(drone.in_air_log())' in the main loop with a break after each iteration of the async for loop.

This is the full error, which occurs when using the "log" methods, or when sending velocity commands:

`
Traceback (most recent call last):
  File "/home/lachlan/Documents/simulator/drone.py", line 211, in in_air_log
    async for in_air in self.drone.telemetry.in_air():
  File "/home/lachlan/Documents/MAVSDK-Python/mavsdk/telemetry.py", line 2891, in in_air
    async for response in in_air_stream:
  File "/home/lachlan/.local/lib/python3.8/site-packages/aiogrpc/utils.py", line 138, in __anext__
    return await asyncio.shield(self._next_future, loop=self._loop)
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/lachlan/.local/lib/python3.8/site-packages/aiogrpc/utils.py", line 126, in _next
    return next(self._iterator)
  File "/home/lachlan/.local/lib/python3.8/site-packages/grpc/_channel.py", line 416, in __next__
    return self._next()
  File "/home/lachlan/.local/lib/python3.8/site-packages/grpc/_channel.py", line 706, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Socket closed"
	debug_error_string = "{"created":"@1593378107.142226325","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1056,"grpc_message":"Socket closed","grpc_status":14}"
>
`

Where I might get either the above exception message, or these two:

`
<_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "failed to connect to all addresses"
	debug_error_string = "{"created":"@1593375936.746898702","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3962,"referenced_errors":[{"created":"@1593375936.644941596","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":394,"grpc_status":14}]}"
>`

`<_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Connection reset by peer"
	debug_error_string = "{"created":"@1593377629.114989693","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1056,"grpc_message":"Connection reset by peer","grpc_status":14}"
>>

`

Now I am very new to MavSDK, so what am I doing wrong? Is there another way of going about this? At the moment my program, might run for a couple minutes or a few seconds before crashing.

JonasVautherin, 2020-06-29:
> StatusCode.UNAVAILABLE

This suggests that `mavsdk_server` is not running. Did you manage to run simple examples, e.g. following [this tutorial](https://auterion.com/getting-started-with-mavsdk-python/)?

lachlanbrewster, 2020-06-29:
I have yes, I used the examples as a way to build my code off of. If it is relevant I am using px4 sitl with Gazebo. I will find out soon enough but I think I found what is wrong. 
After checking asyncio.Task.all_tasks() I found that most, if not all of my tasks created by loop.run_until_complete() were not completing and were stuck on pending. After moving as much of my functionality as I could to the 'background tasks' this error went away. Is there potentially a limit to the number of connections or tasks?

JonasVautherin, 2020-06-30:
> were not completing and were stuck on pending

Something like `drone.telemetry.position()` never ends. Were you expecting it to finish? It's an infinite stream of events that you need to consume. And therefore, probably you want to run that in a separate task.

But something like `drone.action.arm()` will return almost immediately, so you can just call `await drone.action.arm()` as you would expect.

Does that make sense, or do you think you found a bug?

lachlanbrewster, 2020-06-30:
Yes that makes sense, and no I'm fairly sure this was just due to inexperience dealing with MavSDK and doing something strange that caused this. 
Everything seems to work now, the way that I had set up my main loop and background tasks was the problem. Thank you for your help.

closed 2020-06-30 by lachlanbrewster