mavlink/MAVSDK-Python #230: Unable to arm simulation drone

AlexandraBarrett24, 2020-07-07:
>>> from mavsdk import System
>>> drone = System()
>>> await drone.connect('localhost')
>>> await drone.action.arm()
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\aioconsole\execute.py", line 87, in aexec
    result, new_local = await coro
  File "<aexec>", line 2, in __corofn
  File "C:\Users\user\AppData\Local\Programs\Python\Python38-32\lib\site-packages\mavsdk\action.py", line 252, in arm
    raise ActionError(result, "arm()")
mavsdk.action.ActionError: NO_SYSTEM: 'No System'; origin: arm(); params: ()

julianoes, 2020-07-07:
It looks like the connection step did not work, otherwise it would not say `NO_SYSTEM`.

Is this running on the same computer as jMAVSim is running?

AlexandraBarrett24, 2020-07-07:
Yes, this is the same terminal

JonasVautherin, 2020-07-07:
> await drone.connect('localhost')

This is wrong. `connect` takes the MAVLink system address, so something like `udp://:14540`. If you want to connect to `mavsdk_server` running locally (not the embedded one), it has to be done in the line before:

```python
drone = System(mavsdk_server_address="localhost", port=50051)
```

closed 2020-07-11 by AlexandraBarrett24