mavlink/MAVSDK-Python #231: Running examples/camera.py gives error

540KJ, 2020-07-07:
Running `python3 camera.py` in examples gives me this error:

```
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 5283920058631409231
Setting mode to 'PHOTO'
Traceback (most recent call last):
  File "camera.py", line 62, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "camera.py", line 25, in run
    await drone.camera.set_mode(Mode.PHOTO)
  File "/home/jack/Documents/reinforce-3d/venv/lib/python3.6/site-packages/mavsdk/camera.py", line 1708, in set_mode
    mode.translate_to_rpc(request.mode)
TypeError: translate_to_rpc() takes 1 positional argument but 2 were given
```

julianoes, 2020-07-08:
Thanks for the issue. We'll look into it.

What version of mavsdk do you have installed?
```
pip3 show mavsdk
````

540KJ, 2020-07-08:
```
Name: mavsdk
Version: 0.10.0
Summary: Python wrapper for MAVSDK
Home-page: https://github.com/mavlink/MAVSDK-Python
Author: None
Author-email: None
License: UNKNOWN
Location: /home/jack/Documents/reinforce-3d/venv/lib/python3.6/site-packages
Requires: importlib-resources, protobuf, grpcio, aiogrpc
```

julianoes, 2020-07-09:
Ok, thanks. Looks like that's a real bug then that we need to fix.

JonasVautherin, 2020-07-09:
That's [here](https://github.com/mavlink/MAVSDK-Python/blob/master/mavsdk/camera.py#L1708):

```python
async def set_mode(self, mode):
        request = camera_pb2.SetModeRequest()
        mode.translate_to_rpc(request.mode)
```

Any idea why it complains that `translate_to_rpc` is getting 2 positional arguments?

540KJ, 2020-07-09:
https://github.com/mavlink/MAVSDK-Python/blob/b938f1bbf6059cea33d66ef3b5747ad3efc6d780/mavsdk/camera.py#L31

`mode.translate_to_rpc()` should be sufficient?

Except, now when I call `await drone.camera.set_mode(Mode.PHOTO)` from `examples/camera.py`, I get a `CameraError` DENIED. 

https://github.com/mavlink/MAVSDK-Python/blob/b938f1bbf6059cea33d66ef3b5747ad3efc6d780/mavsdk/camera.py#L1691

`mode` parameter never seems to be used meaningfully (i.e. the return value of mode.translate_to_rpc() is never used)?

JonasVautherin, 2020-07-09:
Oh, right, I got it wrong. I believe it used to not return anything but to change the `request.mode` in place. And we changed it to return something.

> mode.translate_to_rpc() should be sufficient?

Something like below, then?

```python
request.mode = mode.translate_to_rpc()
```

Would you mind trying that? Otherwise I'll test it myself ASAP, but that will certainly wait until next week.

540KJ, 2020-07-10:
Added `request.mode = mode.translate_to_rpc()`

Now, `response = await self._stub.SetMode(request)` returns

```
Response camera_result {
  result: RESULT_DENIED
  result_str: "Denied"
}
```

Would this mean that there's something wrong on my end, or that it's still the code?

JonasVautherin, 2020-07-10:
> Would this mean that there's something wrong on my end, or that it's still the code?

I believe that "DENIED" is a response coming from the autopilot (the only place where I see it is [here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/camera/camera_impl.cpp#L543)). So to me it looks like your SITL returns "DENIED", indeed 😇.

PR #232 (2020-07-15): Fix enum translation
Fixing bug reported by @540KJ in #231. Enums were not translated correctly from gRPC, resulting in a runtime crash.

This effectively changes the following:

```python
gimbal_mode.translate_to_rpc(request.gimbal_mode)
```

becomes 

```python
request.gimbal_mode = gimbal_mode.translate_to_rpc()
```

and 

```python
mode.translate_to_rpc(request.mode)
```

becomes

```python
request.mode = mode.translate_to_rpc()
```

@540KJ: would you mind testing this?

Resolves #231.
closed 2020-07-15 by julianoes