mavlink/MAVSDK-Python #233: goto_location results in the aircraft from descending to 0m alt and crashing

iwishiwasaneagle, 2020-07-17:
I have tried the following multiple times, with varying altitudes and airframes (Iris and Plane). However each time this is called, the UAV seemingly just descends until it lands.

```python
#!/usr/bin/env python3

import asyncio
from mavsdk import System, mission_raw

async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break
    
    await drone.action.goto_location(55.8688660,-4.2851267,40,0)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

System:
 - Ubuntu 18.04 LTS
 - Gazebo 9
 - PX4 master

JonasVautherin, 2020-07-17:
That's interesting :sweat_smile:. I think it was used successfully for a multicopter from Java. Are you sure that SITL is not just losing data link and hence doing RTL?

Because your script above will send the `goto` command and terminate, effectively terminating the MAVLink connection. Can you try to add `await asyncio.sleep(20)` after `await drone.action.goto_location(...)` to confirm that it still lands in that case?

iwishiwasaneagle, 2020-07-17:
Yeah no dice even with `await asyncio.sleep(60)`... only log message I see is `INFO [Commander] Freefall detected` until it lands. I tried calling the `goto` during 3 different "states": during a mission, at the end of a completed mission, during a hold after manually flying the iris with a xbox controller.

JonasVautherin, 2020-07-17:
I just tried your code above with `px4-gazebo-headless`, checking the status from QGC:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:latest
```

It seemed like it was working (at least it was flying north-west from Zurich :laughing:).
Would you mind trying it there, too? Just to check what happens when we use the same SITL version.

iwishiwasaneagle, 2020-07-17:
That seems to work (I'm not waiting to see if it flys from Zurich to Glasgow hahaha!) in the docker container, but still not in a fresh VM with the most recent PX4/Firmware.

JonasVautherin, 2020-07-21:
Hmm ok, so apparently something changed in PX4. Question now is whether that's a bug in PX4 or in MAVDSK.

iwishiwasaneagle, 2020-07-21:
What's changed?

julianoes, 2020-07-22:
> `INFO [Commander] Freefall detected`

:astonished: that doesn't sound good! Let me try to reproduce...

julianoes, 2020-07-22:
This works for correctly me against PX4 master.

Can you share a log where it crashes for you?

iwishiwasaneagle, 2020-07-22:
> This works for correctly me against PX4 master.
> 
> Can you share a log where it crashes for you?

Setup:

- Ubuntu 18.04 LTS
- PX4 master
- goto.py example you've just added https://github.com/mavlink/MAVSDK-Python/pull/234

I run the PX4 SITL with `make px4_sitl gazbo_plane` (after building with `DONT_RUN=1 make px4_sitl gazbo_plane`), then run `python3 goto.py`. Below are the relevant logs:

Something else to note is that I saw the same behavior when running
```python
pos = offboard.PositionNedYaw(1000,0,0,0)
log.debug(f"Beginning offboard control: {pos}")
await drone.offboard.set_position_ned(offboard.PositionNedYaw(0.0, 0.0, 0.0, 0.0))
await drone.offboard.start()
await drone.offboard.set_position_ned(pos)
```

```
$ make px4_sitl gazbo_plane
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[2/4] No install step for 'sitl_gazebo'
[3/4] Completed 'sitl_gazebo'
[3/4] cd /vagrant/Firmware/build/px4_sitl_default/tmp && /vagrant/Firmware/Tools/sitl_run.sh /vagrant/Firmware/build/px4_sitl_default/bin/px4 none gazebo plane none /vagrant/Firmware /vagrant/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /vagrant/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: plane
world: none
src_path: /vagrant/Firmware
build_path: /vagrant/Firmware/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/vagrant/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/vagrant/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH :/vagrant/Firmware/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
not running gazebo gui
SITL COMMAND: "/vagrant/Firmware/build/px4_sitl_default/bin/px4" "/vagrant/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/vagrant/Firmware"/test_data
INFO  [px4] Creating symlink /vagrant/Firmware/ROMFS/px4fmu_common -> /vagrant/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_1030
[param] parameter file not found, creating eeprom/parameters_1030
  SYS_AUTOCONFIG: curr: 0 -> new: 1
* SYS_AUTOSTART: curr: 0 -> new: 1030
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 1311244
  CAL_GYRO_PRIME: curr: 0 -> new: 1311244
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  COM_ARM_EKF_AB: curr: 0.0022 -> new: 0.0050
[... 90 lines cut ...]
INFO  [commander] Takeoff detected
INFO  [runway_takeoff] #Takeoff airspeed reached
INFO  [runway_takeoff] #Climbout
INFO  [runway_takeoff] #Navigating to waypoint
INFO  [commander] Landing detected
INFO  [commander] DISARMED by Auto disarm initiated
INFO  [logger] closed logfile, bytes written: 2512406
INFO  [ecl/EKF] 68956000: reset velocity to GPS
INFO  [ecl/EKF] 68956000: reset position to GPS
WARN  [ecl/EKF] 68956000: GPS fusion timeout - reset to GPS
```

```
$ python3 goto.py 
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
Drone discovered with UUID: 5283920058631409231
Waiting for drone to have a global position estimate...
Global position estimate ok
-- Arming
-- Taking off
```

Flight log: https://review.px4.io/plot_app?log=abaf850e-45f7-4084-b250-b089ba2ad1af

julianoes, 2020-07-22:
Ok, thanks for all that info, that's very helpful. I had only tried `iris` before, not `plane`. I did a few tries and `plane` sometimes seemed to do the right thing and other times get too low and crash. For `iris` it now doesn't work anymore but just seems to go sideways at a weird angle, and keep going forever. It's very odd, I'm looking into it.

iwishiwasaneagle, 2020-07-22:
> Ok, thanks for all that info, that's very helpful. I had only tried `iris` before, not `plane`. I did a few tries and `plane` sometimes seemed to do the right thing and other times get too low and [...]

Yeah that sums up the behavior nicely... I'm looking forward to hearing what the root cause was.

julianoes, 2020-07-22:
Ok, I think the `iris` case actually makes sense. It goes the right direction but it does ignore the altitude. It just stays at whatever it was before which is wrong.

For `plane` it seems to work, however, you need to set the altitude correctly. The default SITL takeoff altitude in SITL is 488m, and the goto interface is in AMSL, so you need something > 488m.

iwishiwasaneagle, 2020-07-22:
Ah good to know that it uses AMSL. It's probably worth stating that in the docs seeing as usual absolute altitude is used

hamishwillee, 2020-07-22:
@julianoes @iwishiwasaneagle  The [docs/reference](https://mavsdk.mavlink.io/develop/en/api_reference/classmavsdk_1_1_action.html#classmavsdk_1_1_action_1a6fd615e5571d6e7e3c53a79d2160ffc5) do state that this is an absolute altitude, but the example does not.

FWIW IMO this API definition is a bug - the altitude should be relative to home, at very least "by default" (ideally we should even offer a terrain option). My reasoning is:
1. MAVSDK is supposed to be a simplification of MAVLINK - humans work in "above ground" frame, so forcing everyone to code in addition of ground is a pain. 
2. As a safety issue if you make the default relative to home, then if people don't read, then the vehicle won't crash.

Thoughts?

julianoes, 2020-07-23:
@hamishwillee goto is based on https://mavlink.io/en/messages/common.html#MAV_CMD_DO_REPOSITION which unfortunately is specified in absolute altitude.

If we wanted to use relative altitude we would have to either:
1. Use the reposition flags or reserved param to indicate relative vs. absolute altitude. The problem here is that previous versions of PX4 would ignore that flag and land/crash.
2. Do the calculation based on home position. This would work, just requires the home position to be known before sending a goto. I can see how this would be safer.

JonasVautherin, 2020-07-23:
> Do the calculation based on home position. This would work, just requires the home position to be known before sending a goto. I can see how this would be safer.

This makes sense to me. Would this imply adding a new action result, e.g. `RESULT_HOME_UNKNOWN`?

hamishwillee, 2020-07-24:
So I like option 2, and yes, you'd want a new result(s)

But ideally while I would _default_ it to using home-based altitude I would not _restrict_ it to home because if terrain becomes available a lot of people will prefer that (and some might want AMSL). So if you can have a default and use home for that would be ideal.

And yes, you'd then probably have RESULT_HOME_UNKNOWN or RESULT_TERRAIN_UNKNOWN returned based on the type used.

julianoes, 2020-07-24:
> This makes sense to me. Would this imply adding a new action result, e.g. RESULT_HOME_UNKNOWN?

I think so, yes.

> But ideally while I would default it to using home-based altitude I would not restrict it to home because if terrain becomes available a lot of people will prefer that (and some might want AMSL). So if you can have a default and use home for that would be ideal.

Fair enough, that would be nice.

I think we'll have to rename/change the existing function anyway though, otherwise existing users that use absolute altitude might suddenly be surprised.

I think we could change it to:
1. `goto_location_absolute_altitude()` and `goto_location_relative_altitude()`
2. ```
   enum AltitudeType {
       AboveHomePosition,
       AbsoluteAltitude,
       AboveTerrain
   };
   goto_location(..., AltitudeType altitude_type)
   ```

JonasVautherin, 2020-07-24:
If it's just a matter of preferences, I like 1 better :innocent::

> 1. goto_location_absolute_altitude() and goto_location_relative_altitude()

hamishwillee, 2020-07-27:
`goto_location_absolute_altitude()`, `goto_location_relative_altitude()` and later `goto_location_terrain_altitude()` OK by me.

iwishiwasaneagle, 2020-07-27:
> `goto_location_absolute_altitude()`, `goto_location_relative_altitude()` and later `goto_location_terrain_altitude()` OK by me. Though perhaps `goto_location_home_rel_altitude()` - I've never particularly liked the use of "rel" with the thing that it is relative to not specified.

`goto_location_home_rel_altitude` is more intuitive than `goto_location_relative_altitude`

julianoes, 2020-07-27:
Ok, good point @hamishwillee and @iwishiwasaneagle. In this case it I would suggest `goto_location_home_relative_altitude` :smiley:.

iwishiwasaneagle, 2020-07-27:
Will `goto_location` be removed or maybe add a decorator like `@deprecated` [link](https://stackoverflow.com/a/40301488/4211488)?

julianoes, 2020-07-27:
We might remove it because we haven't marked the API as stable yet, as it's not yet 1.0.0 (but hopefully soon).

iwishiwasaneagle, 2021-07-19:
Getting this off my `issuse` log

closed 2021-07-19 by iwishiwasaneagle