mavlink/MAVSDK-Python #237: mission_raw example.

moonmotor, 2020-07-26:


moonmotor, 2020-07-26:
Has anyone used mission_raw in Python to upload parsed mission items that might have an example to share?
Thanks!

moonmotor, 2020-07-28:
thanks @julianoes for adding the proper label. Do you know if anyone has actually implemented mission_raw in Python? I saw your post about it being available in all languages, but I understand it's relatively new and untested. We're trying to implement a basic mission with mission_item code below, but upon upload, we get a param error indicating one of the parameters is incorrect. If this function is known unsupported, we'll look for an alternate route and abandon mission_raw, but it would be great to be able to issue MAVLink commands.
Thanks for your inputs!

[image]

DoppleGangster, 2021-01-12:
Any updates on this? I am also looking to use the raw command to perform the MAV_CMD_DO_ORBIT. All attempts at mission uploads result in 'INVALID_ARGUMENT' being returned. One item to note is that for params 5 and 6 on the orbit command it states that NaN will use the current position center. These parameters however are both int32_t's which causes an error. 

Thanks for any help I can get!

moonmotor, 2021-01-12:
We were forced to go back to Dronekit. MAVSDK has VERY limited functionality and is riddled with bugs. We wasted a lot of time trying to update to the “better” platform. Hope they can get their act together. Would like to be using Python 3.x for other peripherals.

JonasVautherin, 2021-01-12:
> MAVSDK has VERY limited functionality and is riddled with bugs. We wasted a lot of time trying to update to the “better” platform. Hope they can get their act together.

@moonmotor: Thanks for the constructive feedback, it always feels heartwarming after spending so much free time (on top of the consequent investment from our employers) on an open source project :blush:!

I hope that the reason why I cannot find or remember bug reports (or any kind of contribution) from you is not because of the atmosphere we create around the project. I can ensure you that we do our best to be welcoming and to support users and contributors, even when it involves spending evenings and weekend (in our free time) for that. And I am sorry to read that you wasted time because of our free work, it was not our intention.

If, by any chance, you remember a few of those many bugs and can afford wasting a bit more time sharing them (in a reproducible way), I'm sure we will be happy to have a look, share ideas about how to fix them, and possibly even spend a Saturday afternoon doing it for you (though sometimes we ask those who report bugs to test our fixes, I hope that's fine).

JonasVautherin, 2021-01-12:
@DoppleGangster: It's very difficult for me to say without having code to reproduce the issue. Would you mind opening a new issue (I don't feel like answering to this one anymore) with a meaningful title (maybe "MissionRaw upload results in 'INVALID_ARGUMENT"?), sharing your setup and a small script reproducing your error? That would help, I think :+1:.

moonmotor, 2021-01-12:
It’s coming back to me now. We had the same problem and each argument has to be correct (obviously), but the documentation is not clear and even contrary as to what each argument should be. It took many man-days to get it right, but we did get it right after some time. I’ll check my notes and see if I can find it.

DoppleGangster, 2021-01-13:
I changed my constants to include decimal places where the values required floats. This changed the error to unsupported. 

Looks like I'll be writing my own orbit code.

Thanks for the responses!

JonasVautherin, 2021-01-13:
> but the documentation is not clear and even contrary as to what each argument should be

So the [documentation](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/mission_raw.py#L84-L128) says:

```
    """
     Mission item exactly identical to MAVLink MISSION_ITEM_INT.
     Parameters
     ----------
     seq : uint32_t
          Sequence (uint16_t)
     frame : uint32_t
          The coordinate system of the waypoint (actually uint8_t)
     command : uint32_t
          The scheduled action for the waypoint (actually uint16_t)
     current : uint32_t
          false:0, true:1 (actually uint8_t)
     autocontinue : uint32_t
          Autocontinue to next waypoint (actually uint8_t)
     param1 : float
          PARAM1, see MAV_CMD enum
     param2 : float
          PARAM2, see MAV_CMD enum
     param3 : float
          PARAM3, see MAV_CMD enum
     param4 : float
          PARAM4, see MAV_CMD enum
     x : int32_t
          PARAM5 / local: x position in meters * 1e4, global: latitude in degrees * 10^7
     y : int32_t
          PARAM6 / y position: local: x position in meters * 1e4, global: longitude in degrees *10^7
     z : float
          PARAM7 / local: Z coordinate, global: altitude (relative or absolute, depending on frame)
     mission_type : uint32_t
          @brief Mission type (actually uint8_t)
     """
```

To me that's a one-to-one relationship with the MAVLink message [`MISSION_ITEM_INT`](http://mavlink.io/en/messages/common.html#MISSION_ITEM_INT).

Are you saying that you were using the correct arguments (i.e. generating a valid MAVLink message), but that MAVSDK was failing because of some bug, or are you saying that you had trouble figuring out how to actually use the MAVLink message?

In the former case, an idea would have been to share the failing code, such that we could have fixed the bug (that's how a community works: you get the work of others for free, and when you find a bug, you help reproducing it). In the latter case, maybe we should make it clearer in the documentation that the `mission_raw` plugin, just like `mavlink_passthrough`, is a lower-level abstraction made for more advanced users... Would it have saved you from trying it? The `mission` plugin is indeed more limited, but has a higher-level abstraction, which is convenient when you can't handle the MAVLink messages directly.

moonmotor, 2021-01-13:
Thanks for telling me how a community works – I was unclear about that. I posted several things with no response. You also keep referring to the firmware as “free”. Anything that takes dozens of man-hours to make work isn’t free in my opinion. The MAVADK website touts it as the greatest new API for Mavlink, used in “hundreds of enterprise applications” (or something to that effect), and that’s a load of crap. We couldn’t even connect to the drone using your documentation. By pure trial and error, we found we needed to add a triple slash to the USB port connection string. That was the first bug we found after dozens of wasted hours. That begs the question: who are these hundreds of companies using MAVSDK if nobody has even been able to connect over USB telemetry? It’s one thing to say we’re a group of guys trying to write an API, it’s another to pretend you’re a sophisticated group with a working API.
Whatever – keep your little comments to yourself. I’ll try to DG because it sounds like he’s gonna keep struggling with MAVPOS.

moonmotor, 2021-01-13:
Digging trough my old notes now. I think I got so frustrated with MAVPOS that I threw my notebook away, but I’ll keep looking through emails and trying to remember what the problem was. I remember one problem was that sequence could not start with zero – had to start at 1. But I’ll send what I can find.

moonmotor, 2021-01-13:
OK, DG I think this is an upload that worked.
Not sure if seq can start with zero or has to be 1. Param4 has to be floast(nan) and for some really odd reason the cords have to be nine-digit integers.
Hope this helps. MM

(MissionItem(
1,                     #seq
                             6,                      #frame
                               16,                     #command
                             0,                      #current
                              1,                      #autocontinue
                              0,                     #param1
                              10,                      #param2
                              0,                      #param3
                              float('nan'),           #param4
                              402603986                #param5
                              -754579572,             #param6
                              20,                      #param7
                              0))                     #mission_type

JonasVautherin, 2021-01-13:
> By pure trial and error, we found we needed to add a triple slash to the USB port connection string. That was the first bug we found after dozens of wasted hours.

I'm assuming that the [documentation](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/system.py#L62-L68) of `System.connect()` is not clear enough:

```
        system_address: str
            The address of the remote system. If None, it will
            default to udp://:14540. Supported URL formats:
                - Serial: serial:///path/to/serial/dev[:baudrate]
                - UDP: udp://[bind_host][:bind_port]
                - TCP: tcp://[server_host][:server_port]
```

Or did you start `mavsdk_server` manually? 

```
% ./mavsdk_server -h
Usage: ./mavsdk_server [-h | --help]
       ./mavsdk_server [-p mavsdk_server_port] [Connection URL]

Connection URL format should be:
  Serial: serial:///path/to/serial/dev[:baudrate]
  UDP:    udp://[bind_host][:bind_port]
  TCP:    tcp://[server_host][:server_port]
```

But that's not exactly a bug, rather a documentation issue. Would you have suggestions about improving that?

> That begs the question: who are these hundreds of companies using MAVSDK if nobody has even been able to connect over USB telemetry?

I think that those who fail to connect generally ask us after a few minutes, either on GitHub or on Slack. That's the kind of error we spot in seconds. Again, sorry if we failed to answer in that case.

> I remember one problem was that sequence could not start with zero – had to start at 1.

You mean for the mission items? There is a difference in the MAVLink implementation between Ardupilot and PX4, I think. That's one of the issues of the `mission_raw` plugin unfortunately: you have to figure out the MAVLink implementation before using it. The `mission` plugin abstracts that away. Again, not a bug, rather a low-level protocol definition (MAVSDK has nothing to do with that).

> for some really odd reason the cords have to be nine-digit integers.

Protocol definition, again, see [here](http://mavlink.io/en/messages/common.html#MISSION_ITEM_INT). The [`mission`](https://github.com/mavlink/MAVSDK-Python/blob/main/mavsdk/mission.py#L20-L24) plugin abstracts this away, again:

```
     Parameters
     ----------
     latitude_deg : double
          Latitude in degrees (range: -90 to +90)
     longitude_deg : double
          Longitude in degrees (range: -180 to +180)
```

> I’ll try to help DG because it sounds like he’s gonna keep struggling with MAVPOS.

@DoppleGangster opened an issue [here](https://github.com/mavlink/MAVSDK-Python/issues/297), that's the place for further answers on that regard.

I don't think we'll get anything constructive out of this thread, let me close it. If you have suggestions regarding how we could fix those documentation bugs you mentioned, by all means, please open issues or PRs :pray:. Note that the documentation is generated from the proto files [here](https://github.com/mavlink/MAVSDK-Proto). If that's unclear, feel free to describe your proposed changes in an issue, and we can create the corresponding pull requests.

closed 2021-01-13 by JonasVautherin

moonmotor, 2021-01-13:
Uhhhh…who do you think found the serial:/// bug? I see you updated your documentation to reflect the bug I found. Good for you buddy boy!

I don't think we'll get anything constructive out of this thread, let me close it. If you have suggestions regarding how we could fix those documentation bugs you mentioned, by all means, please open issues or PRs 🙏. Note that the documentation is generated from the proto files here<https://github.com/mavlink/MAVSDK-Proto>. If that's unclear, feel free to describe your proposed changes in an issue, and we can create the corresponding pull requests.