mavlink/MAVSDK-Python #238: How to use MAVSDK in separated scripts?

marlonsecundo, 2020-07-28:
I am having a issue when a running a telemetry script and then a mission script.

Sometimes the scripts execute perfectly. But occasionally the mission script stay stuck in the message:

```
Connected to mavsdk_server!
``` 

MAVSDK support separated connections?

telemetry.py

```py
#!/usr/bin/env python3

import asyncio
from mavsdk import System
import json

async def run():
    # Init the drone
    drone = System()
    await drone.connect(system_address="udp://:14540")

    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    # Start the tasks
    asyncio.ensure_future(print_battery(drone))
    asyncio.ensure_future(print_gps_info(drone))
    asyncio.ensure_future(print_in_air(drone))
    asyncio.ensure_future(print_position(drone))

async def print_battery(drone):
    async for battery in drone.telemetry.battery():
        print(f"Battery: {battery.remaining_percent}")

async def print_gps_info(drone):
    async for gps_info in drone.telemetry.gps_info():
        print(f"GPS info: {gps_info}")

async def print_in_air(drone):
    async for in_air in drone.telemetry.in_air():
        print(f"In air: {in_air}")

async def print_position(drone):

    data = {"position": {}}

    async for position in drone.telemetry.position():
        data["position"]["latitudeDeg"] = position.latitude_deg
        data["position"]["longitudeDeg"] = position.longitude_deg
        data["position"]["absoluteAltitude"] = position.absolute_altitude_m
        data["position"]["relativeAltitude"] = position.relative_altitude_m

        json_data = json.dumps(data)
        print(json_data)

if __name__ == "__main__":
    # Start the main function
    asyncio.ensure_future(run())

    # Runs the event loop until the program is canceled with e.g. CTRL-C
    asyncio.get_event_loop().run_forever()

```

mission.py

```py
#!/usr/bin/env python3

import asyncio

from mavsdk import System
from mavsdk.mission import MissionItem, MissionPlan

async def run():

    drone = System()
    await drone.connect(system_address="udp://:14540")

    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    running_tasks = []
    termination_task = asyncio.ensure_future(observe_is_in_air(drone, running_tasks))

    await drone.mission.clear_mission()
    await drone.action.set_takeoff_altitude(20)
    await drone.action.arm()
    await drone.action.takeoff()
    await asyncio.sleep(20)
    await drone.action.transition_to_fixedwing()
    await drone.action.goto_location(-5.842418, -35.197584, 20, float("nan"))
    await asyncio.sleep(45 * 1)
    await drone.action.return_to_launch()
    # await drone.gimbal.set_roi_location(-5.842418, -35.197584, 490)
    # await drone.action.transition_to_multicopter()
    # await drone.action.land()

    await termination_task

async def observe_is_in_air(drone, running_tasks):
    """ Monitors whether the drone is flying or not and
   returns after landing """

    was_in_air = False

    async for is_in_air in drone.telemetry.in_air():
        if is_in_air:
            was_in_air = is_in_air

        if was_in_air and not is_in_air:
            for task in running_tasks:
                task.cancel()
                try:
                    await task
                except asyncio.CancelledError:
                    pass
            await asyncio.get_event_loop().shutdown_asyncgens()

            return

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())

```

JonasVautherin, 2020-08-03:
> MAVSDK support separated connections?

No, that's a limitation in MAVSDK-C++: when you subscribe to a callback in MAVSDK-C++, you actually replace the older callback. Which means that if you subscribe twice to `mavsdk_server`, only the most recent callback stays.

The solution would be to change the behavior in MAVSDK-C++, such that you can subscribe more than once.

marlonsecundo, 2020-08-04:
OK! Thanks for the explanation! :)

closed 2020-08-04 by julianoes