mavlink/MAVSDK-Python #241: Unable to get Magnetometer information

RogerAylagas, 2020-08-06:
Hi all,

I am trying to get the information from the magnetometer using the following line of code:

`async for compass_info in drone.telemetry.MagneticFieldFrd():
            print(compass_info.forward_gauss)`

but I am getting the following error

> AttributeError: 'Telemetry' object has no attribute 'MagneticFieldFrd'

Can you help me please?

Thank you in advance

julianoes, 2020-08-06:
Could it be called like this?

```
async for compass_info in drone.telemetry.magnetic_field_frd():
   print(compass_info.forward_gauss)
```

RogerAylagas, 2020-08-06:
Thank you for your quick response.

I already tried that but I am getting the same error

> AttributeError: 'Telemetry' object has no attribute 'magnetic_field_frd'

julianoes, 2020-08-06:
Odd. I can reproduce it.

julianoes, 2020-08-06:
Oh, I figured it out:
```
    async for imu in drone.telemetry.imu():
        print(imu.magnetic_field_frd.forward_gauss)
```

closed 2020-08-06 by julianoes

RogerAylagas, 2020-08-06:
Oh thank you so much!! Just one last question how do I convert the magnetometer measurements in Gauss to a compass angle? I cannot find the way of converting it.

Thank you

julianoes, 2020-08-07:
If you need the compass angle, why don't you just use the yaw angle which the PX4 estimator already calculates?

```
    async for euler in drone.telemetry.attitude_euler():
        print(euler.yaw_deg)
```

GestoerteBaecker, 2020-10-09:
> Oh, I figured it out:
> 
> ```
>     async for imu in drone.telemetry.imu():
>         print(imu.magnetic_field_frd.forward_gauss)
> ```

I was faced the exact issue as mentioned by OP, but those lines of code didn't help as well. After executing the script seems to stay in an infinte loop, because there were no ouput in command prompt. On the other hand the code 
```
    async for euler in drone.telemetry.attitude_euler():
        print(euler.yaw_deg)
```
is executed just fine.
I was trying to connect my script to PX4 via USB on Windows 10.

julianoes, 2020-10-22:
Thanks for the heads-up.