mavlink/MAVSDK-Python #243: StatusCode.UNIMPLEMENTED in  await drone.manual_control.start_position_control()

hrnbot, 2020-08-10:
Exception <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.UNIMPLEMENTED
        details = ""
        debug_error_string = "{"created":"@1597055314.543722476","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1062,"grpc_message":"","grpc_status":12}"
>
Result <class 'mavsdk.manual_control.ManualControlResult'>
-- Stopping offboard
Stopping offboard mode failed with error code:               COMMAND_DENIE

here is my python code

`#!/usr/bin/env python3

import asyncio

from mavsdk import System
from mavsdk.offboard import (OffboardError, VelocityBodyYawspeed)
from mavsdk.manual_control import ManualControlError,ManualControlResult,ManualControl

async def run():
    """ Does Offboard control using velocity body coordinates. """

    drone = System(mavsdk_server_address="localhost")
    await drone.connect()
    
    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("-- Arming")
    await drone.action.arm()

    print("-- Setting initial setpoint")
    await drone.offboard.set_velocity_body(
        VelocityBodyYawspeed(0.0, 0.0, 0.0, 0.0))

    print("-- Starting offboard")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed with error code: \
              {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm()
        return

    print("take_off")
    await drone.offboard.set_velocity_body(
        VelocityBodyYawspeed(0.0, 0.0, -1.5, 0.0))
    await asyncio.sleep(5)

    try:
        await drone.manual_control.start_position_control()
    except Exception as mce:
        print("Exception",mce)
    finally:
        print("Result",ManualControlResult)

    print("-- Stopping offboard")
    try:
        await drone.offboard.stop()
    except OffboardError as error:
        print(f"Stopping offboard mode failed with error code: \
              {error._result.result}")

    await drone.action.land()
    
if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
`

JonasVautherin, 2020-08-10:
This means that your `mavsdk_server` does not yet have the `manual_control` implementation. How did you install MAVSDK-Python? From sources? You need [`mavsdk_server` 0.29.0](https://github.com/mavlink/MAVSDK/releases/tag/v0.29.0) at least, and it could be that the latest MAVSDK-Python on pip (`pip install mavsdk`) does not include it yet. It will come with the next release.

hrnbot, 2020-08-11:
I have mavsdk_server 0.29.0 of aarm64 architecture and built the mavsdk from source of e33b5d5c45d129c398512f036d8e67ff40d838ab this commit

JonasVautherin, 2020-08-11:
Just to be sure: the rest of MAVSDK-Python works (takeoff, land, etc), but only `manual_control` fails. Correct?

hrnbot, 2020-08-13:
@JonasVautherin  yes, other than manual control everything working perfectly.

JonasVautherin, 2020-08-13:
Oh, the `manual_control` plugin is not enabled, actually (it should be e.g. here: https://github.com/mavlink/MAVSDK/blob/v0.29.0/src/backend/src/grpc_server.h). Would you be willing to fix that? It should be enough to just look at how other plugins are instantiated and do it similarly 👍.

hrnbot, 2020-08-14:
What I need to change in MAVSDK server also I am currently using arm64 architecture server?

JonasVautherin, 2020-08-14:
e.g. before the `mission` plugin [here](https://github.com/mavlink/MAVSDK/blob/v0.29.0/src/backend/src/grpc_server.h#L27-L28), you would add `manual_control`:

```c++
#include "plugins/manual_control/manual_control.h"
#include "manual_control/manual_control_service_impl.h"
```

Then if you follow `mission`, you'll see it appear in other places, like [here](https://github.com/mavlink/MAVSDK/blob/v0.29.0/src/backend/src/grpc_server.h#L71-L72) or [here](https://github.com/mavlink/MAVSDK/blob/v0.29.0/src/backend/src/grpc_server.cpp#L31). That's how we enable a plugin in `mavsdk_server`, and it needs to be done for `manual_control`.

hrnbot, 2020-08-14:
@JonasVautherin i had done all the changes required. How to build MAVSDK Server for arm64.

closed 2020-08-25 by julianoes