mavlink/MAVSDK-Python #245: standard_init_linux.go:211: exec user process caused "exec format error"

rehadiri, 2020-08-16:
Hello,

I am trying to install Mavsdk on Raspberry pi 4 Raspberry OS Buster Lite. I have installed and checked the docker. It works fine . After that i entered the following command

docker run --rm dockcross/linux-armv7 > ./dockcross-linux-armv7

It installed the dockcross/linux-armv7 image on the docker
But at the end i have got "standard_init_linux.go:211: exec user process caused "exec format error"

I have looked at the dockcross-linux-armv7 in the MAVSDK folder . Its empty , 0 bytes.

What can go wrong here ?
Thanks
Best Regards
Reha

rehadiri, 2020-08-16:
By the way i am following "Building SDK from Source" procedure ( https://mavsdk.mavlink.io/develop/en/contributing/build.html ) . I have successfully build For my laptop running Ubuntu bionic LTS 18.04 . It works fine.

But when i come to Raspberry pi this procedure says that there is a need for cross compilation and we need to use docker for it.

But unfortunately i can't install the docker version.

JonasVautherin, 2020-08-17:
This is a dockcross issue, not related to MAVSDK. This said, are you trying to run dockcross from the RPi? You should run it from your laptop.

rehadiri, 2020-08-17:
Yes i know, its not about MAVSDK. I wonder to know if somebody else has successfully installed MAVSDK on rpi 4 using docker.

JonasVautherin, 2020-08-17:
Dockcross is here for "cross-compiling", i.e. you compile __from__ your laptop, __for__ your rpi. If you want to compile from the rpi for the rpi, you don't need dockcross, you just run a normal compilation.

Does that make sense?

rehadiri, 2020-08-17:
Thanks Jonas, 

I am not very familiar building executables.

Actually before i have tried building the MAVSDK  in Rpi. I did same steps what i did the  for ubuntu.

when i run   **cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_BACKEND=ON -Bbuild/default -H.** 

i have got the following error.log ;

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_87fe2/fast"
/usr/bin/make -f CMakeFiles/cmTC_87fe2.dir/build.make CMakeFiles/cmTC_87fe2.dir/build
make[1]: Entering directory '/home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_87fe2.dir/CheckSymbolExists.c.o
/usr/bin/cc   -fPIE   -o CMakeFiles/cmTC_87fe2.dir/CheckSymbolExists.c.o   -c /home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_87fe2
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_87fe2.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_87fe2.dir/CheckSymbolExists.c.o  -o cmTC_87fe2
/usr/bin/ld: CMakeFiles/cmTC_87fe2.dir/CheckSymbolExists.c.o: in function `main':
CheckSymbolExists.c:(.text+0x48): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_87fe2.dir/build.make:87: cmTC_87fe2] Error 1
make[1]: Leaving directory '/home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_87fe2/fast] Error 2

File /home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_a2f45/fast"
/usr/bin/make -f CMakeFiles/cmTC_a2f45.dir/build.make CMakeFiles/cmTC_a2f45.dir/build
make[1]: Entering directory '/home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_a2f45.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create -fPIE   -o CMakeFiles/cmTC_a2f45.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.13/Modules/CheckFunctionExists.c
Linking C executable cmTC_a2f45
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_a2f45.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_a2f45.dir/CheckFunctionExists.c.o  -o cmTC_a2f45 -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_a2f45.dir/build.make:87: cmTC_a2f45] Error 1
make[1]: Leaving directory '/home/pi/MAVSDK/build/default/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_a2f45/fast] Error 2
  
And i thought that i should go to for docker version. 

I think i did a mistake somewhere but i cant figure out.

JonasVautherin, 2020-08-17:
> /usr/bin/ld: cannot find -lpthreads

There is apparently an issue linking pthreads.

> And i thought that i should go to for docker version.

Well you could, but you have to run dockcross from your desktop computer (Ubuntu 18.04) and not from the RPi. Also I don't think that the RPi 4 is running on armv7, is it? Because above you mention cross-compiling for armv7:

> docker run --rm dockcross/linux-armv7 > ./dockcross-linux-armv7

rehadiri, 2020-08-17:
Hello Jonas,

Yes thank you again  , i could built mavsdk backend using dockcross from my Ubuntu without any problem. I have got the **mavsdk_server** executable on Ubuntu .  I have copied this binary to RPi and the mavsdk_server process successfuly opened. I installed the python-mavsdk by using pip3 install mavsdk. The version 0.11.0 successfuly installed on RPi.  

My sitl and simulator is on Ubuntu. The example code mission.py in RPi can connect to mavsdk_server . But afterwards it can not connect to the jmavsim sitl.

Below is the screen output  ( for mavsdk_server )

pi@raspberrypi:~**/MAVSDK $ ./mavsdk_server -p 50051**
[08:16:16|Info ] MAVSDK version: v0.29.0-3-g2e95299c (mavsdk_impl.cpp:27)
[08:16:16|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:404)
[08:16:16|Info ] Server started (grpc_server.cpp:43)
[08:16:16|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:44)
[08:16:16|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:22) 

for mission.py 

pi@raspberrypi:~ $ python3 mavsdk_mission.py
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...

On the other hand same code ( mission.py ) running on my windows laptop can connect to sitl on the Ubuntu. 
RPi and Windows have the same drone connection script line ( await drone.connect(system_address="udp://:14550")

Sorry for my new problem :)

Thanks Again

rehadiri, 2020-08-17:
My RPi operation system is Raspberry OS Buster Lite 32 bit Os.

rehadiri, 2020-08-17:
I think i understand this problem. When mavsdk and sitl run on the same machine sitl is using the port 14540. But If it is remote ( sitl and mavsdk on different machines ) the sitl broadcast for the port 14550 . 

If we can start the mavsdk_server using the port 14550 , i think this will work . 
We say this in the python script by (  await drone.connect(system_address="udp://:14550" ) . But i afraid this expression is working independently.

JonasVautherin, 2020-08-21:
Because you start `mavsdk_server` manually, `system_address="udp://:14550"` will have no effect. You have to do this when you start `mavsdk_server`:

```
pi@raspberrypi:/MAVSDK $ ./mavsdk_server -p 50051 udp://:14550
```

I hope this helps!

rehadiri, 2020-08-29:
Yes, thanks Jonas

closed 2020-08-30 by JonasVautherin