mavlink/MAVSDK-Python #247: How can I use rospy.Subscriber in MAVSDK-Python code?

HoJae-Lee1, 2020-08-24:
Hi. I'm trying to combine codes of MAVSDK and MAVROS.
I'm trying to put code like
`rospy.Subscriber('mavros/local_position/pose', PoseStamped, cnt.posCb)`
in MAVSDK-Python code such as takeoff_and_land.py to use topics from outside.

Is there any way or examples that I can do so?
Thanks.

julianoes, 2020-08-24:
Usually people use one of the two: ROS with MAVROS, or MAVSDK. Using both doesn't make much sense in my opinion. Why are you using MAVSDK?

HoJae-Lee1, 2020-08-24:
> Usually people use one of the two: ROS with MAVROS, or MAVSDK. Using both doesn't make much sense in my opinion. Why are you using MAVSDK?

Because I need to use topics(information such as target's GPS) from outside. Is there such function in MAVSDK equivalent to MAVROS subscriber?

julianoes, 2020-08-24:
I don't know what you mean with outside? Basically, MAVROS should give you any MAVLink information, including GPS, right?

HoJae-Lee1, 2020-08-26:
I need to communicate with computers other than px4 and ground station.
To do this in MAVROS, you just simply make nodes and subscribe to the topic.
How can I do similar work in MAVSDK then?

That's why I'm trying to mix MAVROS and MAVSDK.

julianoes, 2020-08-26:
You can just use MAVSDK in your ROS node.

closed 2020-08-26 by julianoes