mavlink/MAVSDK-Python #249: > But since I have to manually specify the connection address when i start mavsdk_server

21shrinit, 2020-08-27:
> But since I have to manually specify the connection address when i start mavsdk_server
the second line doesn't do anything?

Good question. You always need both lines, but if `mavsdk_server_address` is specified in the first one, then `system_address` is ignored in the second one. In your case it would mean:

```python
drone = System(mavsdk_server_address='localhost', port=50051)
await drone.connect()
```

And `mavsdk_server` is started manually on 'localhost' with:

```
./mavsdk_server -p 50051 serial:///dev/ttyACM0:57600
```

_Originally posted by @JonasVautherin in https://github.com/mavlink/MAVSDK-Python/issues/189#issuecomment-629145662_

21shrinit, 2020-08-27:
I have tried running mavsdk server manually by following the instructions but it fails to connect please help in this issue

./mavsdk_server -p 50051 serial:///dev/ttyACM0:57600

[02:40:05|Info ] MAVSDK version: v0.29.0 (mavsdk_impl.cpp:27)
[02:40:05|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:404)
[02:40:05|Info ] Server started (grpc_server.cpp:43)
[02:40:05|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:44)
[02:40:05|Info ] Waiting to discover system on serial:///dev/ttyACM0:57600... (connection_initiator.h:22)
[02:40:05|Error] open failed: No such file or directory (serial_connection.cpp:86)
[02:40:05|Error] Connection failed: Connection error (connection_initiator.h:48)

JonasVautherin, 2020-08-27:
As said in the log you posted, it fails [here](https://github.com/mavlink/MAVSDK/blob/develop/src/core/serial_connection.cpp#L86). I think it's because `/dev/ttyACM0` does not exist on your system. Does it?

21shrinit, 2020-08-27:
@JonasVautherin don't know much about /dev/ttyACM0 . I'm using ubuntu 18.04 for simulating px4 with ros gazebo
I tried installing mavsdk with pip3 my python version is 3.6.9 but it was giving error to build and start server manually 
so I followed the instructions and now it is giving this error

JonasVautherin, 2020-08-27:
`/dev/ttyACM0` is for a serial connection. If you just use SITL, you should be able to connect with `udp://:14540`

```
./mavsdk_server -p 50051 udp://:14540
```

21shrinit, 2020-08-27:
It says this 

./mavsdk_server -p 50051 udp://:14540

[06:48:15|Info ] MAVSDK version: v0.29.0 (mavsdk_impl.cpp:27)
[06:48:15|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:404)
[06:48:15|Info ] Server started (grpc_server.cpp:43)
[06:48:15|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:44)
[06:48:15|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:22)
[06:48:15|Error] bind error: Address already in use (udp_connection.cpp:85)
[06:48:15|Error] Connection failed: Bind error (connection_initiator.h:48)

julianoes, 2020-08-27:
>  Connection failed: Bind error

This means something is already running and bound to the port. Check what else you have running.

21shrinit, 2020-08-27:
Thank you for your guidance mavsdk server has started now I'm facing new error while executing mavsdk commands in apython 

from mavsdk import System
>>> drone = System()
>>> await drone.connect()
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
>>> await drone.action.arm()
Traceback (most recent call last):
  File "/home/shrinit/.local/lib/python3.6/site-packages/aioconsole/execute.py", line 87, in aexec
    result, new_local = await coro
  File "<aexec>", line 2, in __corofn
  File "/home/shrinit/MAVSDK-Python/mavsdk/action.py", line 252, in arm
    raise ActionError(result, "arm()")
mavsdk.action.ActionError: NO_SYSTEM: 'No System'; origin: arm(); params: ()

JonasVautherin, 2020-08-27:
How do you know that it detected the drone? `Connected to mavsdk_server` means that MAVSDK is waiting to detect a drone. Say you run `./mavsdk_server udp://:55555` (note the wrong port), you will probably get the same output. Did you use 14540, and are you sure that your SITL broadcasts on 14540?

21shrinit, 2020-08-28:
According to the logs it has mentioned port 14540 along with other ports

roslaunch px4 posix_sitl.launch
... logging to /home/shrinit/.ros/log/8946ed8c-e8f9-11ea-a5a9-88b11135505c/roslaunch-HP-pavilion-ubuntu-18-3872.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://HP-pavilion-ubuntu-18:34515/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    sitl (px4/px4)
    vehicle_spawn_HP_pavilion_ubuntu_18_3872_526664920159043890 (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [3894]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8946ed8c-e8f9-11ea-a5a9-88b11135505c
process[rosout-1]: started with pid [3905]
started core service [/rosout]
process[sitl-2]: started with pid [3908]
process[gazebo-3]: started with pid [3909]
process[gazebo_gui-4]: started with pid [3914]
INFO  [px4] Creating symlink /home/shrinit/Firmware/ROMFS/px4fmu_common -> /home/shrinit/.ros/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
process[vehicle_spawn_HP_pavilion_ubuntu_18_3872_526664920159043890-5]: started with pid [3930]
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] Loaded: eeprom/parameters_10016
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[ INFO] [1598596921.603031015]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1598596921.609883389]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1598596921.613511813]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1598596921.619526021]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[INFO] [1598596921.627963, 0.000000]: Loading model XML from file /home/shrinit/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
[INFO] [1598596921.666776, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1598596924.873170234]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1598596924.905246340]: Physics dynamic reconfigure ready.
[INFO] [1598596924.990025, 0.052000]: Calling service /gazebo/spawn_sdf_model
[INFO] [1598596925.241810, 0.208000]: Spawn status: SpawnModel: Successfully spawned entity
[vehicle_spawn_HP_pavilion_ubuntu_18_3872_526664920159043890-5] process has finished cleanly
log file: /home/shrinit/.ros/log/8946ed8c-e8f9-11ea-a5a9-88b11135505c/vehicle_spawn_HP_pavilion_ubuntu_18_3872_526664920159043890-5*.log
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [commander] Mission #3 loaded, 7 WPs, curr: 0
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: **Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550**
INFO  [mavlink] mode: **Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540**
INFO  [mavlink] mode: **Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030**
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2020-08-28/06_42_06.ulg
INFO  [logger] Opened full log file: ./log/2020-08-28/06_42_06.ulg
INFO  [ecl/EKF] 1384000: GPS checks passed
INFO  [ecl/EKF] 3004000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 3004000: reset position to last known position
INFO  [ecl/EKF] 3004000: reset velocity to zero
INFO  [ecl/EKF] 5884000: reset position to GPS
INFO  [ecl/EKF] 5884000: reset velocity to GPS
INFO  [ecl/EKF] 5884000: starting GPS fusion

21shrinit, 2020-08-28:
Thank you so so much @julianoes and @JonasVautherin thank you for guiding me through these errors. I finally figured it out ,I tried running mavsdk server on port 14550 as it was mentioned in logs and it worked, I got the examples and commands running in simulation. I really appreciate your help as I was struggling with errors from past 3 days.
thank you so much...!!!

JonasVautherin, 2020-08-28:
Nice :rocket:! So it seems like your SITL is not broadcasting on 14540 after all.

Note that an easy way to run SITL (headless) that will have 14540 and 14550 for sure is https://github.com/jonasvautherin/px4-gazebo-headless.

closed 2020-08-28 by JonasVautherin