mavlink/MAVSDK-Python #250: Manual Control Feature

GriffinMack, 2020-09-09:
Hello, I was looking to mess around with the newly added "Manual Control" feature, but noticed that a PyPI release has not been made with this feature included.

Are there any plans for a new PyPI release soon?

JonasVautherin, 2020-09-09:
Good point, I'll make a release later this week :+1:

GriffinMack, 2020-09-09:
@JonasVautherin Thanks! Is there any way an example for manual control could be thrown together before then?

JonasVautherin, 2020-09-10:
MAVSDK-Python 0.12.0 is out: https://github.com/mavlink/MAVSDK-Python/releases/tag/0.12.0. Let us know if it works with manual control!

JonasVautherin, 2020-09-10:
BTW, there is a C++ manual control example [here](https://github.com/mavlink/MAVSDK/tree/develop/examples/manual_control). Would be amazing if you could contribute a PR with an example in python for MAVSDK-Python :blush:

GriffinMack, 2020-09-10:
Awesome! I'll get it downloaded and look into contributing an example if I figure it out. Thanks a lot