mavlink/MAVSDK-Python #257: Calibration - ProgressData contains invalid UTF-8 data

danielghirst, 2020-09-18:
I am using MAVSDK python to do drone calibrations. I am expecting the drone to go through a entire simple calibration using the script attached while printing the status text to the console. What happens is that everything runs as expected up until the accelerator calibration. As soon as the status_text is "Hold still, measuring down side" it immediately give the error: 
```
[libprotobuf ERROR google/protobuf/wire_format_lite.cc:577] String field 'mavsdk.rpc.calibration.ProgressData.status_text' contains invalid UTF-8 data when parsing a protocol buffer. Use the 'bytes' type if you intend to send raw bytes. 
__anext__ grpc exception

```
`#!/usr/bin/env python3

import asyncio
from mavsdk import System
import time 

async def run():

    drone = System()
    await drone.connect(system_address="udp://:14550")
    
    print("-- Starting compass calibration")
    async for progress_data in drone.calibration.calibrate_magnetometer():
        print(progress_data)
    time.sleep(1)

    # await drone.connect(system_address="udp://:14540")
    print("-- Starting gyro calibration")
    async for progress_data in drone.calibration.calibrate_gyro():
        print(progress_data)
    time.sleep(1)
    print("-- Starting accel calibration")
    async for progress_data in drone.calibration.calibrate_accelerometer():
        print(progress_data)
    time.sleep(1)
    print("-- Starting level horizon calibration")
    async for progress_data in drone.calibration.calibrate_level_horizon():
        print(progress_data)
    time.sleep(1)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())`

python 3.6.9
mavsdk 0.12.0
grpcio 1.32.0
grcpio-tools 1.32.0
asyncio 3.4.3
aiogrpc 1.8

julianoes, 2020-09-21:
My guess is that PX4 sounds garbage which is then interpreted as UTF-8.
Do you know if it is "Hold still, measuring down side" that triggers it, or the one that follows afterwards? You can open QGC in parallel to check that.

julianoes, 2020-09-30:
@danielghirst are you running `mavsdk_server` separately or as included in the pip install?

julianoes, 2020-09-30:
And what version of PX4 are you using?

julianoes, 2020-09-30:
Ok, I can reproduce it.

julianoes, 2020-10-01:
Still happens actually.

julianoes, 2021-03-26:
I believe this has been fixed.

closed 2021-03-26 by julianoes