mavlink/MAVSDK-Python #258: To configure behaviour of drone when geofence is violated

21shrinit, 2020-09-21:
examples/geofence.py
In simulation drone continues it's path after violating geofence as obvious the behaviour of drone is not defined.
Can anyone guide me how can i define the behaviour of drone after violation of geofence .I am not able to configure how to catch the warning of geofence violation so i can execute the required behaviour of drone

JonasVautherin, 2020-09-24:
I believe this is done on the PX4 side. So you should be able to set this through parameters, via the `param` plugin.

Does that make sense?

21shrinit, 2020-09-25:
Yes it does makes sense. Can you guide with an example showing how to set parameters as I am not well versed with programming

21shrinit, 2020-09-27:
Thanks I got it how to set param using param plugin .
thank you for your help.

closed 2020-09-27 by 21shrinit