mavlink/MAVSDK-Python #260: Mavsdk_server not starting

Owen-K, 2020-10-01:
I have installed mavsdk for python in a virtual environment using pip3 install mavsdk. When I run the telemetry.py example the program hangs at `Waiting for mavsdk_server to be ready...`. I am using MacOS Catalina as my OS. I have tested the same code on an ubuntu vm and it works as expected. Any suggestions or help debugging is greatly appreciated.

JonasVautherin, 2020-10-01:
Can you find the `mavsdk_server` executable (it got installed by `pip3 install mavsdk` somewhere) and run it manually? That would give you more output to share here :+1:.

Owen-K, 2020-10-01:
When I run `mavsdk_server` which is located in `/usr/local/bin/` it returns the following.
```
mavsdk_server udp://:14550    
[05:41:08|Info ] MAVSDK version: 0.27.0 (mavsdk_impl.cpp:26)
[05:41:08|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:372)
[05:41:08|Info ] Server started (grpc_server.cpp:43)
[05:41:08|Info ] Server set to listen on 0.0.0.0:51875 (grpc_server.cpp:44)
[05:41:08|Info ] Waiting to discover system on udp://:14550... (connection_initiator.h:22)
[05:41:08|Info ] New system on: 192.168.64.3:18570 (udp_connection.cpp:256)
[05:41:08|Debug] Component Autopilot (1) added. (system_impl.cpp:352)
[05:41:09|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:525)
[05:41:09|Info ] System discovered [UUID: 5283920058631409231] (connection_initiator.h:61)
[05:41:09|Warn ] Command ack 520 not matching our current command: 511 (mavlink_commands.cpp:142)
[05:41:09|Warn ] Command ack 511 not matching our current command: 520 (mavlink_commands.cpp:142)
[05:41:09|Warn ] Command ack 520 not matching our current command: 528 (mavlink_commands.cpp:142)
[05:41:09|Debug] MAVLink: critical: Connection to ground station lost (system_impl.cpp:257)
[05:41:09|Debug] MAVLink: info: Data link regained (system_impl.cpp:257)
```

JonasVautherin, 2020-10-01:
> [05:41:08|Info ] MAVSDK version: 0.27.0 (mavsdk_impl.cpp:26)

That's an older version, and I think it came from `brew install mavsdk`. You should be able to update it with brew. But that's not the one that is called by python.

Do you have another one, that would be installed where the `mavsdk` python package is installed?

Owen-K, 2020-10-02:
Here is the output from the mavsdk_server in the python packages directory.
```
mavsdk_server udp://:14550
[09:21:46|Info ] MAVSDK version: 0.27.0 (mavsdk_impl.cpp:26)
[09:21:46|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:372)
[09:21:46|Info ] Server started (grpc_server.cpp:43)
[09:21:46|Info ] Server set to listen on 0.0.0.0:57176 (grpc_server.cpp:44)
[09:21:46|Info ] Waiting to discover system on udp://:14550... (connection_initiator.h:22)
[09:21:46|Info ] New system on: 192.168.64.3:18570 (udp_connection.cpp:256)
[09:21:46|Debug] Component Autopilot (1) added. (system_impl.cpp:352)
[09:21:47|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:525)
[09:21:47|Info ] System discovered [UUID: 5283920058631409231] (connection_initiator.h:61)
[09:21:47|Warn ] Command ack 520 not matching our current command: 511 (mavlink_commands.cpp:142)
[09:21:47|Warn ] Command ack 511 not matching our current command: 520 (mavlink_commands.cpp:142)
[09:21:47|Warn ] Command ack 520 not matching our current command: 528 (mavlink_commands.cpp:142)
[09:21:48|Debug] MAVLink: info: [logger] ./log/2020-10-02/01_21_42.ulg (system_impl.cpp:257)
```

JonasVautherin, 2020-10-02:
hmm that seems normal. What if you run it again, but with `./mavsdk_server -p 50051 udp://:14550`, and connect your Python script to that (instead of letting it start `mavsdk_server` itself), like so:

```python
from mavsdk import System

drone = System(mavsdk_server_address="localhost")
await drone.connect()
```

Does it hang, too? Does `mavsdk_server` react at all, or does it crash or something?

Owen-K, 2020-10-02:
It won't run with the above command and throws an error because the port is in use. It appears that launchd is bound to port 50051.

JonasVautherin, 2020-10-02:
> If I change the port to 50052 it works as expected.

Meaning that it solves your problem? Or that it still fails but running on 50052?

Owen-K, 2020-10-03:
It solves the problem. It appears that apple has a service running on port 50051 causing mavsdk to not be able to use the default port.

JonasVautherin, 2020-10-04:
Good to know!

closed 2020-10-04 by JonasVautherin